nordicsemi/Android-BLE-Library #357: BleManagerGattCallback.onServicesInvalidated() method is not called when peer device disconnects

mecoFarid, 2022-02-10:
`BleManagerGattCallback.onServicesInvalidated()` is called when local device's Bluetooth is turned off. But it is not being called when peer device's Bluetooth is turned off.

Steps to reproduce:
1. Connect to a peer device
2. Turn off peer device's Bluetooth

philips77, 2022-02-10:
Which lib version are you using?
Are you using `.useAutoConnect(true)` in connection request?
Is your device bonded?
Did you wait connection supervision timeout, until the lib gets disconnection event from the system?

mecoFarid, 2022-02-10:
@philips77 I have edited my answer.

I waited more than 3 minutes, repeated the same thing multiple times. Supervision timeout should be 30 seconds if I am not mistaken

mecoFarid, 2022-02-10:
Closing, since this was the iOS peer device's problem. Quoting @philips77 from #356 

> What device are you using as the remote device? Did you try turning it off instead of just turning Bluetooth off? On some devices, e.g. on iOS, disabling Bluetooth in Control Panel just blocks new connections, but doesn't actually turn off Bluetooth for existing devices. Try turning the device off or move away from it.

closed 2022-02-10 by mecoFarid