nordicsemi/Android-BLE-Library #361: Co-existence of multiple

marianhlavac, 2022-02-20:


philips77, 2022-03-16:
> When having multiple managers for a single device, is it possible for the managers to share the connection state (or better re-using the same state machine for all managers)?

No, each manager is associated with a different `BluetoothGatt` object, which is a client object. Disconnecting from one does not mean disconnecting from other.

> ...using enqueue (which is already deprecated btw)

Thanks, will fix that.

> Is there any way to "extend" an existing manager to get more services and characteristics to work within a single manager?

OK, that looks like a feature request for nRF Connect Device Manager library. I'll add it to our backlog. As I understood, you'd like to be able to extend `McuMgrBleTransport` and get access to other services to be able to use them in your app. That's doable, but the change will appear in nRFCDM lib, not here.

philips77, 2022-03-23:
Have a look at https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/pull/44. Would be great to get some feedback from you tomorrow.

closed 2022-06-07 by philips77