nordicsemi/Android-BLE-Library #362: Are there any methods related to queue operation?

jiajunhui, 2022-02-25:
Is there a way to get the length of the current queue and whether there is a way to jump the queue? For example, I want to improve the priority of some commands.

philips77, 2022-03-16:
Hello, in the current version you may only clear the queue using 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5f340654329ec1fd4878ba9ee34c8087ba910116/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1980

jiajunhui, 2022-03-16:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

philips77, 2022-03-21:
Hello, 

I'm thinking of a method to enqueue to the front of the queue. Would that help you? Modification of the order of the already enqueued packets seems dangerous.

philips77, 2022-03-21:
Related: #218

philips77, 2022-03-21:
I was thinking about the feature request. Adding something like `enqueueFirst()` would also imply adding similar methods with `await()` and `suspend()` in the ktx module, but some issues may arise there. Let's say someone is using `suspend()` methods, which end when the request end. If a user then calls `suspendFirst()` or alike, the manager could have already taken next request. I'll need to think more about it.

wm-eisos, 2022-05-04:
> Hello, in the current version you may only clear the queue using
> 
> https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5f340654329ec1fd4878ba9ee34c8087ba910116/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1980

how would i be able to access this function since it is protected ?

LuoPeiQin, 2022-09-15:
This is very useful to me, is there any development plan for this feature now?

philips77, 2022-09-16:
Currently not. May I ask why do you need this method? What is the use case. We may consider changing priorities.