nordicsemi/Android-BLE-Library #363: cannot receive connection state change after removeBond() sometimes

kkkbird, 2022-03-02:
`removeBond()` is called like `removeBond().then(callback).enqueue()` for checking the bond state after remove bond.  and there are two different behaviors after that. 

1. sometimes `Connection state changed` is logged and `onServicesInvalidated` is invoked, `BOND_STATE_CHANGED` is not logged and removeBond callback is not invoked.
```
V/nordic-lib: Removing bond information...
D/nordic-lib: device.removeBond() (hidden)
D/nordic-lib: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
I/nordic-lib: Disconnected
D/nordic-lib: gatt.close()
D/ble-manager: onServicesInvalidated # self-defined ble manager
```

2.  sometimes `BOND_STATE_CHANGED` is received and `onServicesInvalidated` is not invoked.
```
V/nordic-lib: Removing bond information...
D/nordic-lib: device.removeBond() (hidden)
D/nordic-lib: [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I/nordic-lib: Bond information removed
D/nordic-lib: gatt.close()
D/viewmodel: removeBond callback:F2:0F:0E:FB:CD:DF state:10   # self-defined view model
```

in the 1st situation, though the removeBond callback is not invoked, we known the device is disconnected and can reconnect the device as designed. 
in the 2nd situation, because `gatt.close()` is called but `onServicesInvalidated()` is not invoked, we don't known the device is actually disconnected, it leads to some mistake. we have to check connection state again in removeBond callback. I think it should not be a expected result.

after check the source code in nordic ble library, this problem may relate to issue #157. if i remove `close()` call in `BleManagerHandler.java` for #157, logs remain same when 1st situation occurs. but for the 2nd one, `Connection state changed` can be received and `onServicesInvalidated` is invoked, and the connection state can be well handled.
```
V/nordic-lib: Removing bond information...
D/nordic-lib: device.removeBond() (hidden)
D/nordic-lib: [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I/nordic-lib: Bond information removed
D/viewmodel: removeBond callback:F2:0F:0E:FB:CD:DF state:10
D/nordic-lib: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
I/nordic-lib: Disconnected
D/nordic-lib: gatt.close()
D/ble-manager: onServicesInvalidated
```

philips77, 2022-03-02:
Thank you for the report. I'll have a look.

kkkbird, 2022-03-03:
Hi @philips77, thank you. btw, there may be other two questions.
1. if i keep a observer of `ObservableBleManager.bondingState` and it seems that i never receive `NOT_BONDED` state in both situation above. can only get correct value by `BluetoothDevice.getBondState()`
2. for the 1st situation, i set a `then` callback but it is not invoked after `removeBond` request executed, it may lost because of the disconnection but still confusing, as `then` is defined as `will be executed when the request has been processed, no matter the request result.`

philips77, 2022-03-16:
Hello, could you try the proposed solution and check if it works? It's a bit difficult to test it, but you may already have a setup for doing that. Could you place logs from 1. and 2. after applying the PR? Is the behaviour as intended?

closed 2022-03-18 by philips77

philips77, 2022-03-18:
Released in 2.4.0-beta03. Please test if it works for you. We're getting ready to release 2.4.0 final next week.

kkkbird, 2022-03-18:
our office has been temporarily closed since last week because of cvoid-19 :(
hope i can get back to work early next week and i'll test it asap.

kkkbird, 2022-06-07:
hi @philips77 

after 2 month lockdown in shanghai, i finally come back to office to test this issue. 

* 2.3.1: unfortunately, after tried same procedure dozens of times, I found that 2nd situation could not be reproduced on my phone (Samsung S21, Android 12). Maybe it was caused by the system upgrading last month. 
* 2.5.0: the behavior is same as 2.3.1 without `then()` callback being invoked.
```
I/nordic-lib: Disconnected
D/ble-manager: onServicesInvalidated
```
Moreover, system BOND broadcast still seemed to not be handled correctly on 2.5.0, If add a log before  `createBond()` and check the log after `removeBond()` and call `createBond()` again, bond value looked like below:

`ObservableBleManager.bondingState.getValue().getState()` : BONDED
`ObservableBleManager.getBluetoothDevice().getBondState()` : 10 => BOND_NONE

PR #365 (2022-03-18): Making sure then() callback is called for RemoveBond request
This PR fixes #363.

### Issue

When `removeBond()` is executed, there are 2 callbacks called in result: disconnection and changing bond state to `BOND_NONE`. They, however, as shown in #363, can come in any order. Until now, the behaviour was different in any of those cases. 

### Solution

This PR fixes the issue by ensuring that both the `onConnectionStateChange` callback and bond state broadcast listener are called. They still can be called in any order, but neither of them should block execution of the other. The `then` callback [... cut]
PR #571 (2024-08-27): fix: remove_bond request should be handled after disconnect
this PR is about to fix #363 case 1, `this.request` will reset to `null` in call of `nextRequest(false)` below and callback for `removeBond` won't be invoked correctly even though bond status change signal is received later. how ever it may introduce unintended side effects as noted in the comment above. Please help to review it.

https://github.com/NordicSemiconductor/Android-BLE-Library/blob/bfcfdf9f5916ac2b06c9de6ed335b11dc53301c2/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2200-L2212