nordicsemi/Android-BLE-Library #364: Kotlin: confusing and undocumented BleManager.stateAsFlow behaviour

DavidJRobertson, 2022-03-16:
I had assumed that calling BleManager.stateAsFlow() multiple times would return the same flow, but it does not. Further, calling it actually breaks previously-returned flows as it overwrites the connection observer that is used to update the flow. 

Ideally, the same flow should be returned. This could be accomplished by storing the MutableStateFlow instance on the ConnectionObserver subclass, and then only replacing the ConnectionObserver if it is not the right type.

philips77, 2022-03-16:
On it. Thank you for reporting it.

philips77, 2022-03-16:
Hi, could you please review the PR?

closed 2022-03-18 by philips77

philips77, 2022-03-18:
Released in 2.4.0-beta03. Please test if it works for you. We're getting ready to release 2.4.0 final next week.

PR #366 (2022-03-18): Returning the same instance of state observers in ktx
This PR fixes #364.

Both connection and bond state observers can be called multiple times and the same flow instance is returned.
As suggested in the mentioned issue, if a different observer was set, calling `stateAsFlow()` or `bondingStateAsFlow()` will throw.