nordicsemi/Android-BLE-Library #369: Does this support multiple peripherals?

williamtan37, 2022-03-18:
Hi, 

I read the documentation and it says the service supports multiple connections. Does this mean it supports multiple peripherals? Would appreciate a confirmation.

Thanks!

philips77, 2022-03-21:
Yes. A `BleManager` instance can be used to connect to a single device, but you may have multiple instances of the manager, each for a different device. Phone should support at least 6 simultaneous connections, but keep in mind, that all connections share the same controller, so packets are sent in serial, not parallel.

Also. if needed, more than one manager instance can be used to connect to the same device, e.g. to separate some functionality. But then all of the managers keep separate connection state, and i.e. all need to be disconnected for the device to disconnect.

philips77, 2022-03-21:
Related: #361

closed 2022-03-21 by philips77