nordicsemi/Android-BLE-Library #374: Device is not supported issue

dononcharles, 2022-03-28:
Hi,
I got "priority: 5 <==>  message:Device is not supported" error when i was trying to use the lib.
Please i would like to ask if this lib can be used for other Bluetooth LE chip different from NordicSemiconductor Ble chip?
Thanks

philips77, 2022-03-29:
Hello,
Yes, it can be used with any BLE chip. Although we recommend those from Nordic :)

The error you are getting is thrown as your `isRequiredServiceSupported(BluetoothGatt)` method returns *false*. Your implementation failed to find required service among those discovered.
It may be that:
1. Your Android device has cached old services on previous connection.
   
   You may either restart Bluetooth, or install [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp), connect to your device, then select **Refresh services** and then try again in your app.
2.  Your `isRequiredServiceSupported(BluetoothGatt)` implementation is invalid and you return *false*.
   
    Have a look at [example in nRF Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/014316a6fad1ae024fe32f53051b59c5ea911a2c/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java#L170-L185).
3. You're connecting to a wrong device.

    Make sure it the correct device you're trying to connect to.
4. The services on your device are in fact different.

    We recommend the mentioned nRF Connect app, which displays the actual services present on the device.

closed 2022-03-29 by philips77

dononcharles, 2022-03-29:
Thanks for your reply @philips77

dononcharles, 2022-03-29:
@philips77, after carefully read your answer, i can connect to the device. 
But please i need a little bit your light for how to require for pairing. 
Below is the function i use to connect:

`fun connectDevice(currentDevice: BluetoothDevice) {
        clientManager = ClientManager()
        clientManager?.connect(currentDevice)
            ?.useAutoConnect(true)
            ?.retry(3, 100)
             ?.usePreferredPhy(PhyRequest.PHY_LE_1M_MASK or PhyRequest.PHY_LE_2M_MASK)
            ?.timeout(15_000)
            ?.before {
                Logger.warn("CONNECTING : $it")
            }
            ?.done {
                Logger.warn("CONNECTED")
            }
            ?.fail { device, status ->
                Logger.warn("$device is failed with status $status to connect to the controller")
            }
            ?.enqueue()
    }`

How can i make sure the device is bonded before call this function? thx

philips77, 2022-03-29:
You may do this in many ways.
1. Usually, if your device requires bonding, it has some security on at least one characteristic or descriptor (reading or writing to it requires encryption). An attempt to read/write triggers automatic bonding.
2. You may initiate bonding from the Android phone even before that using `createBond()` request - this however, will not work on iOS, if you're planning to do iOS app as well. On iOS 1. is the only way to create bond, and the bonding process is transparent to the app. The response just comes delayed comparing to no-bonding case.

After the device is connected, and your `isRequiredServiceSuppoted(BluetoothGatt)` method returns *true*, the library will call `initialize()` method in your manager. The intention of this method is to initialize your device, e.g. enable notifications, set notification callbacks, request MTU, PHY, etc. See an example [here](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/014316a6fad1ae024fe32f53051b59c5ea911a2c/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java#L161-L167). It may also trigger bonding process by either 1. or 2. 
```kotlin
override fun initialize() {
    // [...]
    createBond()
        .done {
             Logger.warn("BONDED")
        }
        .fail { device, status ->
             Logger.warn("Bonding failed with status $status")
        }
        .enqueue();
    // [...]
    // Here you may also read / write something to a protected characteristic instead of the above.
}
```

philips77, 2022-03-29:
Your
```kotlin
?.done {
    Logger.warn("CONNECTED")
}
```
call will be called AFTER the initialization queue is complete, not only after the device has connected.

dononcharles, 2022-03-29:
Your help is very appreciated. thx
Take a look  on class if you've seen something not correct:

```kotlin
class BluetoothLeService2 : Service(), KoinComponent {

    private var clientManager: ClientManager? = null
    private var binder: IBinder = LocalBinder2()

    fun connectDevice(currentDevice: BluetoothDevice) {
        clientManager = ClientManager()
        clientManager?.connect(currentDevice)
            ?.useAutoConnect(true)
            ?.retry(3, 100)
            ?.timeout(15_000)
            ?.before {
                Logger.warn("CONNECTING : $it")
            }
            ?.done {
                Logger.warn("CONNECTED")
            }
            ?.fail { device, status ->
                Logger.warn("$device is failed with status $status to connect to the controller")
            }
            ?.enqueue()
    }

    fun writeData(payload: ByteArray) {
        clientManager?.sendData(payload)
    }

    inner class ClientManager : BleManager(this@BluetoothLeService2) {

        val bgxServiceUUID: UUID = UUID.fromString(AllGattServices.BGX_STREAMING_SERVICE_UUID)
        val txUUID: UUID = UUID.fromString(AllGattCharacteristics.BGX_TX_CHARACTERISTIC_UUID)
        val rxUUID: UUID = UUID.fromString(AllGattCharacteristics.BGX_RX_CHARACTERISTIC_UUID)

        var txCharacteristic: BluetoothGattCharacteristic? = null // for transmission ( Controller --> App)
        private var rxCharacteristic: BluetoothGattCharacteristic? = null // for sending (App --> Controller)

        override fun getGattCallback(): BleManagerGattCallback {
            return GattCallback()
        }

        override fun log(priority: Int, message: String) {
            if (BuildConfig.DEBUG || priority == Log.ERROR) {
                Logger.warn("priority: $priority <==>  message:$message")
            }
        }

        fun sendData(payload: ByteArray) {
            writeCharacteristic(rxCharacteristic, payload, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)
                .with(rxCallback)
                .enqueue()
        }

        private inner class GattCallback : BleManagerGattCallback() {

            override fun initialize() {
                ensureBond()
                    .done {
                        Logger.warn("BONDED")
                    }
                    .fail { device, status ->
                        Logger.warn("$device: Bonding failed with status $status")
                    }
                    .enqueue()

                // You may enqueue multiple operations. A queue ensures that all operations are
                // performed one after another, but it is not required.
                beginAtomicRequestQueue()
                    .add(requestMtu(247) // Remember, GATT needs 3 bytes extra. This will allow packet size of 244 bytes.
                        .with { device: BluetoothDevice?, mtu: Int -> log(Log.INFO, "$device: MTU set to $mtu") }
                        .fail { device: BluetoothDevice?, status: Int -> log(Log.WARN, "$device: Requested MTU not supported: $status") })
                    .add(enableNotifications(txCharacteristic))
                    .done { device: BluetoothDevice? -> log(Log.INFO, "$device: Target initialized") }
                    .enqueue()

                setNotificationCallback(txCharacteristic).with { device, data ->
                    Logger.warn("$device : ${data.getStringValue(0).toString()}")
                }

                readCharacteristic(txCharacteristic).with(txCallback).enqueue()
            }

            override fun isRequiredServiceSupported(gatt: BluetoothGatt): Boolean {
                val bgxService = gatt.getService(bgxServiceUUID)
                if (bgxService != null) {
                    rxCharacteristic = bgxService.getCharacteristic(rxUUID)
                    txCharacteristic = bgxService.getCharacteristic(txUUID)
                }

                var writeRequest = false
                if (txCharacteristic != null) {
                    writeRequest = txCharacteristic?.properties ?: 0 > 0
                }

                Logger.warn("isRequiredServiceSupported: $writeRequest")
                return rxCharacteristic != null && txCharacteristic != null && writeRequest
            }

            override fun onServicesInvalidated() {
                rxCharacteristic = null
                txCharacteristic = null
            }

        }
    }

    val rxCallback = DataSentCallback { device, data ->
        Logger.warn("DataSentCallback $device, $data")
    }

    inner class LocalBinder2 : Binder() {
        val service: BluetoothLeService2
            get() = this@BluetoothLeService2
    }

    override fun onBind(intent: Intent?): IBinder = binder

    override fun onUnbind(intent: Intent?): Boolean {
        // close service
        return super.onUnbind(intent)
    }

    val txCallback = DataReceivedCallback { device, data ->
        Logger.warn("DataReceivedCallback $device, $data")
    }
}
```

philips77, 2022-03-29:
1. Regarding logging:
    ```kotlin
    override fun log(priority: Int, message: String) {
        if (BuildConfig.DEBUG || priority == Log.ERROR) {
            Logger.warn("priority: $priority <==>  message:$message")
        }
    }
    ```
    I recommend the new method added in 2.4.0-beta02 or 03 `getMinLogPriority()` instead. You may return;
    ```kotlin
    override fun getMinLorPriority() = if (Build.DEBUG) Log.VERBOSE else Log.ERROR
    ```
    This will make the BLE faster on production, as logs will not even be created, not only not logged.
2. In `isRequiredServiceSupported(BluetoothGatt)` you may make use of `.run { }` method, like [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/30d7375c6d2c96eeb7483d94ae64499f2a57f136/profile_prx/src/main/java/no/nordicsemi/android/prx/data/PRXManager.kt#L121), this may be more readable and more kotlin'y.
3. I would move call to `setNotificationCallback(...)` above the `beginAtomicRequestQueue(...)`, so that you don't loose any notification (now you enable them before the callback is set, so with some small chance you may miss one or two, if the connection interval is short.
4. You may also want to use `Flow`. We have added support for that in `ble-ktx` extension. Have a look how the notifications are handled [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/30d7375c6d2c96eeb7483d94ae64499f2a57f136/profile_prx/src/main/java/no/nordicsemi/android/prx/data/PRXManager.kt#L102).

dononcharles, 2022-03-29:
i very thank you. your lib is very super good. I can not connect to my controller, write and read.
i will use your feedbacks to make it more powerful :)

dononcharles, 2022-03-31:
@philips77, everything is working now perfectly until i am getting this below error when disconnect from the device:

BluetoothLeService2 has leaked IntentReceiver no.nordicsemi.android.ble.BleManager$1@1148f6e that was originally registered here. Are you missing a call to unregisterReceiver()?
    android.app.IntentReceiverLeaked: Service BluetoothLeService2 has leaked IntentReceiver no.nordicsemi.android.ble.BleManager$1@1148f6e that was originally registered here. Are you missing a call to unregisterReceiver()?
        at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:1717)
        at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:1494)
        at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1765)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1731)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:1719)

Am I missing something to close? i do not also create any receiver in the BluetoothLeService2 service.
Thx for your help

dononcharles, 2022-03-31:
I just find the solution.
BluetoothLeService2 service class is not using the application or activity context. 
So the context am passing to this inner class  "inner class ClientManager : BleManager(this@BluetoothLeService2)" is not correct.
To solve it i inject the application into this connect function.
```koltin
 fun connectDevice(currentDevice: BluetoothDevice, context: Context) {
        clientManager = ClientManager(context)
       .....
 }

inner class ClientManager(context: Context) : BleManager(context) {
...
}
```