nordicsemi/Android-BLE-Library #379: Blemanager. setMtu does not take effect. How can I solve the problem

jswn0203, 2022-04-20:
Blemanager. setMtu does not take effect. How can I solve the problem

philips77, 2022-04-20:
It depends. Please, provide logs obtained using the 'log(int, String)` method.
It may be that your target device does not support MTU higher than 23. Or that you're not connected. Or, that you're not calling this method. Or, that it does take effect. With no logs I can't say anything.

jswn0203, 2022-04-24:
I call Manger. RequestMtu after I connect to the device or after I receive the data, but onMtuChange has no callback

jswn0203, 2022-04-24:
mtuRequest

philips77, 2022-04-28:
Have a look at a sample code in [nRF Connect Device Manager](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/d377af1c394cd606648fc672cdf21499b91bec56/mcumgr-ble/src/main/java/io/runtime/mcumgr/ble/McuMgrBleTransport.java#L600-L607).

closed 2022-04-28 by philips77