nordicsemi/Android-BLE-Library #381: Handle BleServerManager

pateychn, 2022-04-28:
Hello,

I'm implementing a BleServerManager for a client that needs to read a live computed value.

After reading the following PR : #143, and the library code, I see that onCharacteristicReadRequest is fully handled and returns the current characteristic value.

Is there a way to handle the readRequest and send a live computed value ? I looked at the waitForRead request but I think this is not what I want as the server queue is suspended until the characteristic is read.

philips77, 2022-04-28:
So what you can do, is use 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/42128bf3b0c6b571404992c81df44c21f33f89d2/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1112-L1128
whenever the characteristic value changes. This will set a new value to the characteristic, which can be read any time from a remote device. As you said, the `onCharacteristicReadRequest` is handled automatically, and will return the value set using the above method.

philips77, 2022-04-28:
Instead of returning the current value on read event, you set the value when it changes. The library returns the current value.

Remember, that you may have a shared characteristic (the same value for any connected device), or a dedicated characteristic (each connected device may read a different value). You define it when setting up the server.

closed 2022-04-28 by philips77

pateychn, 2022-04-28:
Thanks for your answer. 

Indeed I'm using setCharacteristicValue, but in my case the client device executes a readCharacteristic to request the current time. 
So I understand that I should set the characteristic value every second. 
Could that have an impact on the operation queue (Maybe not while the client does not perform a read request ?) ?

sundaylol, 2022-06-28:
> 感谢您的回答。
> 
> 实际上，我使用的是 setCharacteristicValue，但在我的情况下，客户端设备执行读取字符来请求当前时间。所以我明白我应该每秒设置一个特征值。这会对操作队列产生影响吗（也许在客户端不执行读取请求时没有影响？）？

Did you find a solution? I also encountered this problem

sundaylol, 2022-06-28:
> Thanks for your answer.
>
> Indeed I'm using setCharacteristicValue, but in my case the client device executes a readCharacteristic to request the current time. So I understand that I should set the [...]

Did you find a solution? I also encountered this problem

philips77, 2022-06-28:
Please, follow #385. We will add a callback to send read response in some future release. You may also contribute and try creating a PR!

philips77, 2022-09-30:
Hello,
I've now completed the task. Could you please give it a try? I will release 2.6.0-alpha02 shortly.