nordicsemi/Android-BLE-Library #382: Getting the device name after connecting the device on Android9.0 and above is null

Fly-Felix, 2022-04-29:
When I run on Android 9.0 and above, the scanned device can get the name, but once I click on the device to select the connection, the device name immediately becomes null after the connection is successful, and I have not found similar problems after checking a lot of information. come here to try your luck

Fly-Felix, 2022-04-29:
Everything works fine when scanning the device, getName() device name returns null after connecting the device

philips77, 2022-04-29:
The `BluetoothDevice.getName()` is obtained from *Device Name* characteristic after connection, and cached by the system. I actually recommend getting the name from the advertising data instead, and keeping it together with the `BluetoothDevice` object. When you, for example, flash a new fw on a DK, with a different application, which is advertising a different name, the `getName()` still returns the old, cached one.

Fly-Felix, 2022-04-29:
I seem to know, I just checked the source code of ExtendedBluetoothDevice in Android-nRF-Toolbox, and found that the constructor ExtendedBluetoothDevice(@NonNull final ScanResult scanResult) was called during scanning and assigned to the name, and there was no name change in it. The subsequent list update, This can achieve the effect of a cache name, I understand it right? , my list adapter in the update directly used the bluetooth device update list for the convenience of use, and reassigned the name, it should be this problem, I will try in this direction, thank you.

Fly-Felix, 2022-04-30:
Good morning, I tried it yesterday, and the name is temporarily saved successfully, but when I reopen the page, I get the list of connected devices through List<BluetoothDevice> devices = manager.getConnectedDevices(BluetoothProfile.GATT); It returns a BluetoothDevice, so the name still cannot be Get , the connected bluetooth device will no longer broadcast, and of course will not enter onScanResult, so the ScanResult and name cannot be obtained.

Fly-Felix, 2022-04-30:
I have experimented with the latest results. After I went to the system to connect the BLE device once, BluetoothDevice.getName() all returned to normal.

Fly-Felix, 2022-04-30:
It seems that the system will not cache the name of the third-party APP of Android 9.0 and above when connecting to a Bluetooth device

philips77, 2022-09-30:
Sorry for the ling response time.
Please, check the name returned by Device Name characteristic in nRF Connect for Mobile. This is the name returned from `BluetootDevice.getName()`.
I'm closing the issue due to long time of inactivity.

closed 2022-09-30 by philips77