nordicsemi/Android-BLE-Library #399: Android 13 beta: Disconnect not working

solita-jtn, 2022-06-08:
I have issue with Android 13 beta. Disconnect doesn't work properly. 

While the library is reporting that the device is disconnected, it is not discoverable and it can be only connected from same Android mobile.

Have you tried about Android 13? Any ideas what might be causing this?

LionZXY, 2022-06-09:
Also similar behaviour, but with Android 12

philips77, 2022-06-27:
This means, that the device is still connected to Android. `BluetoothGatt` or its wrapper `BleManager` are just virtual connections.
Every app can hold multiple virtual connections to the same device. Disconnecting one virtual connection doesn't mean, that the device is disconnected (even if you receive STATE_DISCONNECTED).
It only means, that your single client got disconnected. Android maintains the physical link and closes it when the last client has disconnected. Also, the Android system itself can also be a client, e.g. for some bonded devices. 

To make sure the device gets disconnected you need to request graceful disconnection or reset from the device side by sending some op code. Terminating a connection from the device side closes all virtual connections on Android, as the physical link gets closed.

This is not a but, it's a feature :)

philips77, 2022-06-27:
You may also toggle Bluetooth on the phone, but that's poor used experience.

closed 2022-06-27 by philips77

solita-jtn, 2022-07-01:
As described in the original post the difference was between **13 beta and older version**. Disconnect didn't work in 13 while in older ones it worked. Now it seems that in new version (latest update) for 13 beta the problem has been fixed.

philips77, 2022-09-17:
Hi, did you try it on the final version of Android 13?