nordicsemi/Android-BLE-Library #402: AbstractMethodError thrown in BleManagerHandler.onServicesInvalidated()

ExCx, 2022-06-22:
An IoT device controller library uses this library to communicate with device. After I send the settings, the device resets itself, disconnects and triggers the onConnectionStateChange callback method. Inside, there's this method called notifyDeviceDisconnected, which calls an abstract method called onServicesInvalidated.

That method always throws an AbstractMethodError and crashes my app.

I cannot do anything about it. As it's a triggered event, I can't wrap it in a try-catch block so I may ignore the error and prevent the crash.

What would cause this and is there any way that I can just ignore it some way?

Stack trace is as follows:
```
java.lang.AbstractMethodError: abstract method "void no.nordicsemi.android.ble.BleManagerHandler.onServicesInvalidated()"
        at no.nordicsemi.android.ble.BleManagerHandler.notifyDeviceDisconnected(BleManagerHandler.java:1660)
        at no.nordicsemi.android.ble.BleManagerHandler.access$900(BleManagerHandler.java:59)
        at no.nordicsemi.android.ble.BleManagerHandler$3.onConnectionStateChange(BleManagerHandler.java:1905)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:223)
        at android.app.ActivityThread.main(ActivityThread.java:7660)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

vasuthakker, 2022-07-04:
Any update on this issue ?
How to resolve this ?

sylwester-zielinski, 2022-07-04:
Hello,

Thank you for reporting an issue.

Recently a signature of the `onServicesInvalidated()` method has changed and it has been made abstract which means that you should override this method in `BleManagerGattCallback` that you are returning in your overriden `BleManager.`.

Please double check that you have this method. 

I suspect that your runtime version of the library may be different than compile time.
So please ensure that you have the same library version in all your gradle modules. The best if you can update everything to the latest one.

vasuthakker, 2022-07-04:
Thank you for your response,

I'll check it from my end now..

closed 2022-09-29 by philips77