nordicsemi/Android-BLE-Library #403: How does Gattserver onCharacteristicReadRequest call back variables

sundaylol, 2022-06-28:
I need to call back a variable in Gattserver onCharacteristicReadRequest. How should I do it here? Is there any sample code? Can newWaitForWriteRequest only set constants to wait for reading? It was very hard for me to check the source code here for a day, and I really have no idea, so I came to ask for advice.

philips77, 2022-06-28:
Please, follow #385.

Currently, the only way is to use one of the following:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/52d60a4de110b73faef05938ea31a699a9a38163/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1115-L1226
to set a value whenever the characteristic value changes.
The read request will automatically be replied with the current value of the characteristic.

However, I understand that you may not want to set the value constantly, so a callback approach would be better. We created #385 as a task to implement this feature, but this will have to wait until after summer.

closed 2022-06-28 by philips77

sundaylol, 2022-06-29:
> Please, follow #385.
>
> Currently, the only way is to use one of the following: [...]

Thanks for the reply, I will try to modify it myself first