nordicsemi/Android-BLE-Library #407: Example with asFlow()?

AndreKR, 2022-07-10:
Starting from the example code in the readme I am able to get notifications by changing the `initialize()` method to
```kotlin
        override fun initialize() {
            requestMtu(517).enqueue()

            setNotificationCallback(realTimeDataC).with { device, data ->
                Log.i("CT", "Notofication received")
            }
            enableNotifications(realTimeDataC).enqueue()
        }
```

It is unclear to me how I can get a flow with these notifications.

I tried adding a method like this to MyBleManager:
```kotlin
    fun realTimeDataFlow(): Flow<Data> {
        val flow = setNotificationCallback(realTimeDataC).asFlow()
        enableNotifications(realTimeDataC).enqueue()
        return flow;
    }
```

This didn't work, nothing is collected on the flow. I suspect this is maybe because the method is called _before_ initialize?

If that is in fact the reason, then I don't see how I could make it work. The flow is needed when the activity starts and `initialize()` is only called some time later.

AndreKR, 2022-07-10:
As a workaround, I'm sending the data through a channel now:
```kotlin
        override fun initialize() {
            requestMtu(517).enqueue()

            setNotificationCallback(realTimeDataC).with { device, data ->
                realTimeDataChannel.trySend(data)
            }
            enableNotifications(realTimeDataC).enqueue()
        }
```

philips77, 2022-07-14:
Have a look at nRF Toolbox, for example here:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/100a98d0328177aa7138a0c8d065b11b2ea6ba1f/profile_csc/src/main/java/no/nordicsemi/android/csc/data/CSCManager.kt#L94-L126

philips77, 2022-07-14:
Indeed, your `realTimeDataFlow()` must be called when the device is ready (initialize complete) or in `initialize`. Before that, the characteristic references are null.

closed 2022-09-29 by philips77