nordicsemi/Android-BLE-Library #409: example for splitWithProgressFlow()？

almosting, 2022-07-19:


corentin-c, 2022-09-15:
Providing a description would be nice, with what you want to do with it for example
Something like this should work

`
    suspend fun writeCharacteristicWithProgression(
        characteristic: BluetoothGattCharacteristic,
        data: ByteArray,
        writeType: Int,
        splitter: DataSplitter
    ): Flow<ProgressIndication> {
        return withContext(Dispatchers.IO) {
            val writeRequest = super.writeCharacteristic(characteristic, data, writeType)
            val flow = writeRequest.splitWithProgressFlow(splitter)
            writeRequest.enqueue()
            flow
        }
    }
`

philips77, 2022-09-16:
Thank you @corentin-c !

almosting, 2022-09-26:
Thank you @corentin-c

closed 2022-09-26 by almosting