nordicsemi/Android-BLE-Library #411: disconnect the device

Kevin-Leung, 2022-07-21:
hi, can u explain to disconnect a connected device, 
I'm trying to use disconnect() in 
public final DisconnectRequest disconnect() {
		return Request.disconnect().setRequestHandler(requestHandler);
	}
but nothing happend,

niklasdahlheimer, 2022-07-21:
In your class which extends BleManager you can call

```java
disconnect().enqueue();
```

which triggers an async disconnect. you can add callbacks when disconnect succeeds or fails by calling

```java
  disconnect()
                .done(this::onDisconnected)
                .fail(this::onConnectionOperationFailed)
                .enqueue();
```
where onDisconnected and onConnectionOperationFailed are functions in your class which have a signature like
```kava
protected void onDisconnected(BluetoothDevice device){
// your callback code
}
```

philips77, 2022-07-24:
Thank you @niklasdahlheimer for excellent answer!

closed 2022-07-24 by philips77