nordicsemi/Android-BLE-Library #414: setPreferredPhy in initialize() fails on Android 13

bobatsar, 2022-08-26:
Setting a preferred PHY in the `initialize()` method, fails on Android 13 (on a Pixel 4a) but works on older Android versions.

`onDeviceReady()` is never called on Android 13.

Setting the preferredPHY in the `connect` request doesn't seem to work at all, even on older devices.
```kotlin
 clientManager.connect(device)
    .useAutoConnect(false)
    .usePreferredPhy(PhyRequest.PHY_LE_2M_MASK)
    .enqueue()
```

On older devices, I read the PHY in onDeviceReady() and it displays 2M only when setting it in `initialize()` not via the `connect`request.

I adapted the `GattService.kt` example to work with my BLE device (uses Nordic UART service).
The Android 13 version receives the notifications but `onDeviceReady()` is not called.

I am testing with 2.5.1 on the latest commit 52d60a4de110b73faef05938ea31a699a9a38163

```kotlin

override fun onDeviceReady() {
    Log.i(TAG, "--------------------------------------")
    Log.i(TAG, "device is ready")
    Log.i(TAG, "--------------------------------------")
    readPhy().enqueue()
    super.onDeviceReady()
}

override fun initialize() {

    setNotificationCallback(mTXCharacteristic)
        .with { _, data ->
            if (data.value != null) {
                val value = String(data.value!!, Charsets.UTF_8)
                defaultScope.launch {
                    myCharacteristicChangedChannel?.send(value)
                }
            }
        }

    enableNotifications(mTXCharacteristic).enqueue()

    if (true && Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
        val bluetoothManager =
            context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager
        if (bluetoothManager.adapter?.isLe2MPhySupported == true) {
            setPreferredPhy(
                PhyRequest.PHY_LE_2M_MASK,
                PhyRequest.PHY_LE_2M_MASK,
                PhyRequest.PHY_OPTION_NO_PREFERRED
            ).enqueue()
        } else {
            Log.w(TAG, "PHY LE 2M not supported. Run with default 1M")
        }
    }
}
```

philips77, 2022-08-26:
Thank you for creating the issue. We will investigate it, hopefully next week.

philips77, 2022-08-26:
> Setting the preferredPHY in the connect request doesn't seem to work at all, even on older devices.

Yes.. I know, this is kind of a dead API. Android just prefers PHY 2LE if the other side supports it and that's all.

philips77, 2022-08-26:
Similarly, scanning with LE_CODED only doesn't work and always returns both 1M and Coded (if the phone can scan coded at all).

philips77, 2022-08-26:
Also, here:
```kotlin
        val bluetoothManager =
            context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager
        if (bluetoothManager.adapter?.isLe2MPhySupported == true) {
            setPreferredPhy(
                PhyRequest.PHY_LE_2M_MASK,
                PhyRequest.PHY_LE_2M_MASK,
                PhyRequest.PHY_OPTION_NO_PREFERRED
            ).enqueue()
        } else {
            Log.w(TAG, "PHY LE 2M not supported. Run with default 1M")
        }
```
you don't have to check it, as the library does it for you, and may report `fail` if not supported.

bobatsar, 2022-08-26:
I just checked with the nrf connect app and there also setting the preferred phy doesn't work on the Pixel 4a. There is just a spinning wheel. On another phone with Android 11 nothing happens but I don't know how can I see the currently used PHY in the nrf connect app. 
There is the option to read it but I can't find a way to display it.

philips77, 2022-08-26:
It is printed in the log on INFO level. Swipe the services pane to the right to see the log, or select menu->Show log.

bobatsar, 2022-09-01:
FYI: it seems the new Gabeldorsche bluetooth stack in Android 13 created multiple issues.
https://issuetracker.google.com/issues?q=android%2013%20bluetooth

Could be that this issue is also due to the new stack.

bobatsar, 2022-09-08:
FYI: I just did the Android 13 september update and the issue still persists on the Pixel 4A

weliem, 2022-09-08:
Looks like an Android 13 bug. I did some isolated tests without using this library and onPhyUpdate is simply never called.

...using a Pixel 5

philips77, 2022-09-29:
Here's the sniffer trace from a test:
[image]

Phone used: Pixel 5, Android 13 (September 5, 2022 security update).

philips77, 2022-09-29:
For comparison, here's Huawei Mate 20 Pro (Android 9) with the same test:
[image]

philips77, 2022-09-29:
Tests on Pixel 6 with:
```kotlin
connect(device)
   .usePreferredPhy(PhyRequest.PHY_LE_2M_MASK)
   .enqueue()
```

The preferred PHY was completely ignored:
[image]
despite the device reporting support for it in feature exchange:
[image]

philips77, 2022-09-29:
The same when using Pixel 5:
[image]
1. The preferred PHY is ignored during connection, all packets are sent with PHY 1M
2. Calling `setPreferredPhy` causes disconnection

Fun fact: switching to PHY 2M actually works. The LLCP Length Request packet selected above is sent using PHY 2M:
[image]

philips77, 2022-09-29:
Ok, so what can be done now:
1. We leave it as is, blaming Android, perhaps filing a bug report. This has some changes to be fixed in Android 14, I believe. Perhaps with a security update?
2. We may test other (non-Pixel) phones with Android 13, checking if it's on Android side, or Google part.
3. In BLE library I can add "unless Android 13" check in `setPreferredPhy` to not call it at all. After all, it's "preferred", and "no change" is better than a disconnection.

bobatsar, 2022-09-29:
Thanks for the tests. It seems the Android Bluetooth team was currently working on a fix for the disconnect issue with bonded devices (see https://issuetracker.google.com/issues/242755161) and they said they will ship a fix in an upcoming security update. Perhaps this is also the reason for the disconnect issues on your side.
In my tests (we don't bond the device) I don't have disconnects but as @weliem wrote, the callback is not called so the Android-BLE-Library never completes the connect completely.

So filing a bug report would at least make sure they know of the issue. If you could do that, that would be great. 

Currently there aren't much Android 13 phones available but that should change soon.

Checking for Android 13 is what I am doing now. This would at leas make sure the connection can be handled but of course this should be removed if this bug is fixed on the android side.

philips77, 2022-09-29:
I edited https://github.com/NordicSemiconductor/Android-BLE-Library/issues/414#issuecomment-1262259744

closed 2022-09-29 by philips77

weliem, 2022-09-29:
Indeed just the lack of the callback. To make matters worse, I found out that the callback IS called when a Phy request is not supported! Then I get an error....

philips77, 2022-09-29:
I just merged a PR with a workaround. I'll release a new version, could you give it a spin?

philips77, 2022-09-29:
Version *2.6.0-alpha01* should be public in couple of minutes.

bobatsar, 2022-09-29:
It seems to work on my Pixel 4a. Thanks for the fix @philips77 

When will be an official release?

philips77, 2022-09-29:
My idea was to let people test the new features for couple of days and wait for a feedback. I think there are also 2 other issues I'd like to look into before releasing 2.6.0, but I don't expect any breaking API changes now.

philips77, 2022-09-29:
Btw, in the PR you may see, that there's a 1 second delay, which I give to the native API to call the proper callback. I tried making to 400ms, but that was too short in my case. However, this depends on the connection interval, as setting PHY requires 3 packets.
It may not be enough for all cases. Perhaps would be a good idea to base it on the connection parameters callback.

philips77, 2022-09-29:
Sorry, there was an issue in GitHub Actions workflow and the release failed. I'm releasing 2.6.0-alpha01 on Maven Central now.

bobatsar, 2022-09-30:
I tested now the 2.6.0-alpha01 with my app and have one more issue.
I connect to two devices and the first setPhy on the first device always failes (no error, but the PHY is not updated to 2M.
When I retry with setPhy, it is correctly set to 2M.

I can handle this in my code but perhaps this is a general issue.
Besides that, it is working.

philips77, 2022-10-04:
Could you share some logs? Also, the current version is `2.6.0-alpha03`, but there were no changes to PHY in later releases.

bobatsar, 2022-10-07:
Here are some logs, tested with latest October security update on Pixel 4a.

You can see, that the first `onPhyRead()` returns 1M. When calling `setPreferredPhy()` again, it is working to set 2M. There is even an error message in the log:
```
packages/modules/Bluetooth/system/gd/hci/hci_layer.cc:156 on_command_status: Received UNEXPECTED command status:CONTROLLER_BUSY opcode:0x2032 (LE_SET_PHY)
```

It seems the bluetooth stack also reports the missing callback for `onPhyUpdate`
```
2022-10-06 18:23:08.673  bluetooth               W  packages/modules/Bluetooth/system/main/shim/acl.cc:758 OnPhyUpdate: Dropping ACL event with no callback
```

```
2022-10-06 18:23:06.481  bt_btif_gattc           I  packages/modules/Bluetooth/system/btif/src/btif_gatt_client.cc:365 btif_gattc_open_impl: Transport=2, device type=2, address type =1, phy=1
2022-10-06 18:23:06.487  bt_btif_gattc           I  packages/modules/Bluetooth/system/btif/src/btif_gatt_client.cc:365 btif_gattc_open_impl: Transport=2, device type=2, address type =1, phy=1
2022-10-06 18:23:06.547  bluetooth               I  packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:303 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:7c:03 callback:Connected
2022-10-06 18:23:06.548  BluetoothRemoteDevices  E  aclStateChangeCallback: device is NULL, address=F7:59:0A:2E:7C:03, newState=0
2022-10-06 18:23:06.549                          D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2022-10-06 18:23:06.600  bluetooth               I  packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:303 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:c3:a2 callback:Connected
2022-10-06 18:23:06.608                          D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2022-10-06 18:23:07.430                          V  Requesting preferred PHYs...
2022-10-06 18:23:07.430                          D  gatt.setPreferredPhy(LE 2M, LE 2M, coding option = No preferred)
2022-10-06 18:23:07.771                          V  Requesting preferred PHYs...
2022-10-06 18:23:07.771                          D  gatt.setPreferredPhy(LE 2M, LE 2M, coding option = No preferred)
2022-10-06 18:23:08.434                          W  Callback not received in 1000 ms
2022-10-06 18:23:08.434                          V  Reading PHY...
2022-10-06 18:23:08.435                          D  gatt.readPhy()
2022-10-06 18:23:08.440  BluetoothGatt           D  onPhyRead() - status=0 address=F7:59:0A:2E:7C:03 txPhy=1 rxPhy=1
2022-10-06 18:23:08.440                          I  PHY read (TX: LE 1M, RX: LE 1M)
2022-10-06 18:23:08.446                          I  F7:59:0A:2E:7C:03 PHY Updated: (TX: LE 1M, RX: LE 1M)
2022-10-06 18:23:08.446                          W  F7:59:0A:2E:7C:03 PHY not updated -> retrying 1/5
2022-10-06 18:23:08.533                          V  Requesting preferred PHYs...
2022-10-06 18:23:08.533                          D  gatt.setPreferredPhy(LE 2M, LE 2M, coding option = No preferred)
2022-10-06 18:23:08.537  bluetooth               E  packages/modules/Bluetooth/system/gd/hci/hci_layer.cc:156 on_command_status: Received UNEXPECTED command status:CONTROLLER_BUSY opcode:0x2032 (LE_SET_PHY)
2022-10-06 18:23:08.665  bluetooth               W  packages/modules/Bluetooth/system/main/shim/acl.cc:758 OnPhyUpdate: Dropping ACL event with no callback
2022-10-06 18:23:08.673  bluetooth               W  packages/modules/Bluetooth/system/main/shim/acl.cc:758 OnPhyUpdate: Dropping ACL event with no callback
2022-10-06 18:23:08.774                          W  Callback not received in 1000 ms
2022-10-06 18:23:08.774                          V  Reading PHY...
2022-10-06 18:23:08.774                          D  gatt.readPhy()
2022-10-06 18:23:08.778  BluetoothGatt           D  onPhyRead() - status=0 address=DE:31:9A:5F:C3:A2 txPhy=2 rxPhy=2
2022-10-06 18:23:08.778                          I  PHY read (TX: LE 2M, RX: LE 2M)
2022-10-06 18:23:08.784                          I  DE:31:9A:5F:C3:A2 PHY Updated: (TX: LE 2M, RX: LE 2M)
2022-10-06 18:23:09.537                          W  Callback not received in 1000 ms
2022-10-06 18:23:09.537                          V  Reading PHY...
2022-10-06 18:23:09.537                          D  gatt.readPhy()
2022-10-06 18:23:09.542  BluetoothGatt           D  onPhyRead() - status=0 address=F7:59:0A:2E:7C:03 txPhy=2 rxPhy=2
2022-10-06 18:23:09.542                          I  PHY read (TX: LE 2M, RX: LE 2M)
2022-10-06 18:23:09.547                          I  F7:59:0A:2E:7C:03 PHY Updated: (TX: LE 2M, RX: LE 2M)
2022-10-06 18:23:25.873                          D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2022-10-06 18:23:25.880                          D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2022-10-06 18:23:26.867  l2c_link                W  packages/modules/Bluetooth/system/stack/l2cap/l2c_link.cc:500 l2c_link_timeout: TODO: Remove this callback into bcm_sec_disconnect
2022-10-06 18:23:26.869  l2c_link                W  packages/modules/Bluetooth/system/stack/l2cap/l2c_link.cc:500 l2c_link_timeout: TODO: Remove this callback into bcm_sec_disconnect
2022-10-06 18:23:26.894  bluetooth               I  packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:303 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:7c:03 callback:Disconnected
2022-10-06 18:23:26.895  BluetoothRemoteDevices  E  aclStateChangeCallback: device is NULL, address=F7:59:0A:2E:7C:03, newState=1
2022-10-06 18:23:26.900  bluetooth               I  packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:303 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:c3:a2 callback:Disconnected
```

philips77, 2022-10-07:
Looks like in your case this 1 second is not enough to complete the PHY request, like I described here: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/414#issuecomment-1262763453

What are the connection parameters used for the connection, do you know?

philips77, 2022-10-07:
If you have the BLE lib included as source, could you try increasing the delay from 1 sec to 2+ seconds, for testing purposes?

jose7165, 2023-01-13:
> I edited [#414 (comment)](https://github.com/NordicSemiconductor/Android-BLE-Library/issues/414#issuecomment-1262259744)
> 
> It was the app triggering disconnection. The only issue is lack of `onPhyUpdate(..)` callback.

Hi! I am just wondering if there was ever an Android issue created to capture this issue at Android 13?

bobatsar, 2023-01-26:
@jose7165 I don't think there was an Android Issue created yet.

PR #426 (2022-09-29): Reading PHY if no callback for setting preferred one
This PR fixes #414.

As `onPhyUpdate(..)` callback doesn't get called on Android 13 (see mentioned issue), as a workaround we initiate a *read PHY* operation (giving 1 sec to change its mind). `onPhyRead(..)` callback seems to be working and it is completing the request initiated with `setPreferredPhy(...)`.