nordicsemi/Android-BLE-Library #415: Delay between writing request to the characteristic

georrge1994, 2022-08-29:
Hi, @philips77  and android-ble-lib team. Thank you for you work! Can you give me an advice in my investigation? 

Input data: app send to device two small commands one by one (~20-50 mSec between them). On the android side **done-callbacks** executes to close to each other, but on the device we see 1 second delay between commands (once we see 2 seconds delay - all time the delay is close to integer value). It happen not often, but time to time. 

Hardware team says that they repeated the problem with nRf-connectivity app too. They suggest that problem inside API. I checked lib implementation and found some delay for inner reconnections for example. Not sure, that it links, but maybe you can suggest why it  happen?

I write commands without confirmation:
`        writeCharacteristic(
            deviceCharacteristics[UUID_FITNESS_MACHINE_CONTROL_POINT_CHARACTERISTIC],
            bytes,
            BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE
        ).done {
            log.info("GATT indoor bike, writeCommand was done")
        }.fail { _, status ->
            log.info("GATT indoor bike, writeCommand was failed (status = $status)")
        }.enqueue()`

philips77, 2022-08-30:
Hi,
With `WRITE_TYPE_NO_RESPONSE` write type the data are sent without confirmation from the fw, thus can be sent in a single connection interval. The `done` callback is called immediately when the data are written to the outgoing queue on Android, not actually when they are sent/received. This may take a while and perhaps some repetitions are needed in case of transmission failure. What you get on fw is, I think, the actual timing when the device receives the data.

In order to synchronize the transmitter and receiver you may use `WRITE_TYPE_DEFAULT` (with response) - in this case the `done` callback is called after the device acknowledges receiving the data, but this type of writing is also much slower, as each packet must reach the application on the device, gets ack'ed and only then you can send the next one. Usually writing with response (aka write request) (and indications) are used to send commands, and write without response (aka write command) (or notifications) to send data.

You may also use notifications from the device to notify the sender when the data have been received, but this is application layer logic and needs to be implemented on both sides: mobile app and the firmware on the device.

georrge1994, 2022-08-30:
Thank you for quick answer and ideas for investigation!

closed 2022-08-30 by georrge1994