nordicsemi/Android-BLE-Library #416: connect fail and the staus =2

RiceRichhonour, 2022-09-13:
cant connect success to bluetooth device when i use the  [Android-BLE-Library].  
the fail staus is 2.

```
 BlinkyManager(this).connect(bluetoothDevice!!)
                // Automatic retries are supported, in case of 133 error.
                .retry(3 /* times, with */, 100 /* ms interval */)
                // A connection timeout can be set. This is additional to the Android's connection timeout which is 30 seconds.
                .timeout(10000 /* ms */)
                // The auto connect feature from connectGatt is available as well
                .useAutoConnect(true)
                // This API can be set on any Android version, but will only be used on devices running Android 8+ with
               .timeout(10000 )
               .before {
               }
               .done{
                   showToast("connect success")
               }
               .fail { device, status ->

                   showToast("connect fail"+status)
                }
                .then( {  })
               .enqueue()

```

but when i use 

  ```
 mBluetoothGatt = bluetoothDevice?.connectGatt(this@SearchBlueToothDeviceActivity, false, mBluetoothGattCallback)
            mBluetoothGatt?.connect()
```

i can receive the successful msg.

RiceRichhonour, 2022-09-13:
I has restart Bluetooth, and  install [nRF Connect], connect to your device, then select Refresh services and then try again in my 
 app.
but  It's useless

RiceRichhonour, 2022-09-13:
> I has restart Bluetooth, and install [nRF Connect], connect to your device, then select Refresh services and then try again in my app. but It's useless
th  nRF Connect  app  can connect to the bluetooth device

nirbhaymanoj, 2022-09-14:
Even I am trying to connect but the thing is the device gets connected but within 1 or 2 seconds it gets disconnected showing status error of -2.
Any help would be appriciated.

Code Snippet:

 bleManager.connect(device).
                                useAutoConnect(false).
                                timeout(15_000).
                                retry(3,100).
                                done(device1 -> Log.d(">>>>>","connected to device")).
                                fail((device1, status) -> Log.d(">>>>>","Connect request failed : "+status)).
                                then(device1 -> Log.d(">>>>>","then clause"+device1)).enqueue();

philips77, 2022-09-29:
The error -2 is specified here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/d8c480ade684df940385cdb0da228c7268f38e97/ble/src/main/java/no/nordicsemi/android/ble/callback/FailCallback.java#L32

That means, that this method:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/d8c480ade684df940385cdb0da228c7268f38e97/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1603-L1610
returned *false*. Check your implementation (UUIDs, properties of characteristics, etc.).

If that method returns *false*, you'll get error -2, meaning that the device you want to connect to isn't supported by your app (e.g. has different services).

closed 2022-09-29 by philips77

philips77, 2022-09-29:
Btw, in the other code you mentioned:

```kotlin
mBluetoothGatt = bluetoothDevice?.connectGatt(this@SearchBlueToothDeviceActivity, false, mBluetoothGattCallback)
mBluetoothGatt?.connect()
```
you should not call `connect()` in the second line. The `connectGatt(...)` does the job. Only when your device disconnects and you want to reconnect **with auto connect flag set**, use `gatt.connect()`.