nordicsemi/Android-BLE-Library #417: How can i use notifyConnectionPriorityChanged in ConnectionPriorityRequest.java to change the interval?

spike1212xie, 2022-09-13:
I am learning and try to make a project,which using ble.But now i can only use requestConnectionPriority from BluetoothGatt.java to change interval,i wanna to change interval to 3200,which mean i cant be success.Today I find this lib,so can you tell me how can i use notifyConnectionPriorityChanged?I think its a solution to solve my issue.Because notifyConnectionPriorityChanged  is not public,so cannot be accessed from outside package.

philips77, 2022-09-13:
You need to use `requestConnectionPriority(..)` from the`BleManager`. Only 3 options are possible, the same as when using native API, which the lib is using under the hood.

closed 2022-09-13 by philips77

spike1212xie, 2022-09-14:
Thanks！I have saw it,but i have also use BluetoothGatt in my project,which also include a requestConnectionPriority.So I cant use requestConnectionPriority from the BleManager,or the another way of describing is that I dont know how to use requestConnectionPriority from the BleManager and BluetoothGatt in the same time.

philips77, 2022-09-16:
If you're using the native API, this is the method: https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)
The corresponding method from BleManager:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/52d60a4de110b73faef05938ea31a699a9a38163/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1882-L1886
is calling the native method, so only the same 3 options are available.
Here you have the values for each of the options:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/52d60a4de110b73faef05938ea31a699a9a38163/ble/src/main/java/no/nordicsemi/android/ble/ConnectionPriorityRequest.java#L49-L82,

It is not possible to request a custom values for the connection parameters from Android. Unfortunately. But you can request them from the peripheral side.