nordicsemi/Android-BLE-Library #421: Multiple Bluetooth connections will automatically disconnect after successful connection

123yangu, 2022-09-21:
Why do multiple Bluetooth connections occur at the same time? After successful connection, they are automatically disconnected from Huawei P30 mobile phones

philips77, 2022-09-21:
Sorry, but I don't understand your question.

123yangu, 2022-09-26:
Generally, the maximum number of Bluetooth connections for Android devices is 7. However, when I connect 6 devices at the same time, I will automatically disconnect Bluetooth after successful connection

123yangu, 2022-09-26:
The current processing method is to make the next connection after a successful/failed connection. There is no problem for the time being, but there is a problem. The connection efficiency is too low

123yangu, 2022-09-26:
> Sorry, but I don't understand your question.

Generally, the maximum number of Bluetooth connections for Android devices is 7. However, when I connect 6 devices at the same time, a Bluetooth device will automatically disconnect after successful connection

liujoshua, 2022-12-13:
P30 might be limited to 6. Number of concurrent connections is dependent on device hardware and installed OS

closed 2022-12-14 by philips77