nordicsemi/Android-BLE-Library #422: How to check if Bluetooth is enabled?

schattennarr, 2022-09-21:
Normally I would go over the bluetoothAdapter and check if isEnabled gives true like this:
```java
BluetoothManager bluetoothManager = context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager
BluetoothAdapter bluetoothAdapter = bluetoothManager.getAdapter()
if (bluetoothAdapter.getIsEnabled() == true) {...}
```
But using the BleManager i don't have access to the Adapter and neither BleManager.isReady nor BleManager.isConnected give me information about that.
Do I have to initialise a bluetoothManager just to get these informations?

philips77, 2022-09-21:
Hello,

Yes, the `BleManager` does not return the state of Bluetooth adapter. You need to check it on your own, which, as you pointed out, requires just 3 lines of code.
The purpose of the `BleManager` is mostly to wrap the connection, rather than replace all Bluetooth API.

Also, I don't know how this check could look like. As a static method? It doesn't fit there, to be honest.

schattennarr, 2022-09-22:
Aye thats right. It just feels a bit awkward to initialise two managers but that's how it is. Thanks for the fast answer and have a nice week :)

closed 2022-09-22 by schattennarr