nordicsemi/Android-BLE-Library #435: Best solution for automatic reconnect

Leopold84, 2022-10-29:
Hi @ all!
I have an app that should always connect to a device independently without user action. Now I'm wondering how best to reconnect if the first connection attempt after app start fails? The connect.useAutoConnect(true) flag only works if the first attempt was successful? I can think of 2 solutions: 
1.: Set connect.timeout() to a very high value and hope that the devices ist reachable in this time period.
2.: Start a new connection setup in the connect.fail() callback. Is there a better solution?

philips77, 2022-10-29:
The solution 1 won't work, as there's 30 sec timeout for connecting with autoConnect = false.

To be honest, I think this use case needs modification of the library. Such feature was requested before as well. I'll change the API, so it will be possible to request autoConnect also on the first connection.

However, the correct way to do this is be starting Bluetooth LE scan with `PendingIntent`. You will get a broadcast when the device is found, and you may immediately connect using the current version on BLE library. As the device is most probably still in range, connection with be quick.

philips77, 2022-10-29:
Here's the API:
https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)
Available from Android 8, or emulated using background service in Scanner Compat Library.

Leopold84, 2022-10-29:
Hi Philips77!

Thanks for your answer! I'm not sure if an API change is really necessary? I have now made the following test with the following call and had the desired result: The connection establishment or the reconnect was carried out without any further function calls?

` this.connect(bluetoothDevice)
      .retry(10, 250)  
      .useAutoConnect(true)
      .timeout(24 * 3600 * 1000 )
      .done(onConnectSuccess)
      .fail(onConnectFail)
      .enqueue();`

Here are the logs:

--- APP STARTS AND TRIES TO CONNECT TO ALREADY KNOWN MAC ---

2022-10-29 21:16:52.473 29862-30039/? D/TstMini: enableBtAdapter()
2022-10-29 21:16:52.488 29862-30039/? D/TstMini: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2022-10-29 21:16:52.495 29862-29862/? D/TstMini: Observer: onDeviceConnecting
2022-10-29 21:16:57.529 29862-29862/? D/TstMini: [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2022-10-29 21:16:57.530 29862-29862/? W/TstMini: Error: (0x85): GATT ERROR
2022-10-29 21:16:57.531 29862-29862/? D/TstMini: wait(250)
2022-10-29 21:16:57.783 29862-29862/? D/TstMini: gatt.connect()
2022-10-29 21:16:57.795 29862-29862/? D/TstMini: Observer: onDeviceConnecting

--- DEVICE SWITCHED ON AFTER 50 MINUTES --

2022-10-29 22:06:06.816 29862-29862/? D/TstMini: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2022-10-29 22:06:06.817 29862-29862/? I/TstMini: Connected to A0:76:4E:71:8E:62
2022-10-29 22:06:06.823 29862-29862/? D/TstMini: wait(300)
2022-10-29 22:06:06.824 29862-29862/? D/TstMini: Observer: onDeviceConnected
2022-10-29 22:06:07.126 29862-29862/? D/TstMini: gatt.discoverServices()
2022-10-29 22:06:07.173 29862-29862/? I/TstMini: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2022-10-29 22:06:07.376 29862-29862/? I/TstMini: Services discovered
2022-10-29 22:06:07.381 29862-29862/? D/TstMini: gatt.setCharacteristicNotification(00001a00-0000-1000-8000-00805f9b34fb, true)
2022-10-29 22:06:07.385 29862-29862/? D/TstMini: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2022-10-29 22:06:07.402 29862-29862/? I/TstMini: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2022-10-29 22:06:07.403 29862-29862/? I/TstMini: Notifications enabled
2022-10-29 22:06:07.408 29862-29862/? D/TstMini: gatt.readCharacteristic(00001a00-0000-1000-8000-00805f9b34fb)
2022-10-29 22:06:07.426 29862-29862/? I/TstMini: Read Response received from 00001a00-0000-1000-8000-00805f9b34fb, value: (0x) 30-31-30-31-30-31-30-31
2022-10-29 22:06:07.429 29862-29862/? D/TstMini: gatt.setCharacteristicNotification(00001a01-0000-1000-8000-00805f9b34fb, true)
2022-10-29 22:06:07.432 29862-29862/? D/TstMini: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2022-10-29 22:06:07.435 29862-29862/? D/TstMini: IOs onDataReceived: 01010101
2022-10-29 22:06:07.435 29862-29862/? D/TstMini: onIoStatusCharacteristicRead() IO-STATUS: 01010101
2022-10-29 22:06:07.436 29862-29862/? D/TstMini: onIoStatusCharacteristicRead() IO-STATUS: 01010101
2022-10-29 22:06:07.438 29862-29862/? I/TstMini: Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
2022-10-29 22:06:07.485 29862-29862/? I/TstMini: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2022-10-29 22:06:07.485 29862-29862/? I/TstMini: Notifications enabled
2022-10-29 22:06:07.485 29862-29862/? D/TstMini: gatt.readCharacteristic(00001a01-0000-1000-8000-00805f9b34fb)
2022-10-29 22:06:07.754 29862-29862/? I/TstMini: Read Response received from 00001a01-0000-1000-8000-00805f9b34fb, value: (0x) 30-30-31-31-32-32-33-33-34-34-35-35-36-36-37-37-38-38-39-39-30-30-31-31-32-32-33-33-34-34-35-35-36-36-37-37-38-38-39-39-30-30-31-31
2022-10-29 22:06:07.754 29862-29862/? D/TstMini: FDS onDataReceived: 00112233445566778899001122334455667788990011
2022-10-29 22:06:07.754 29862-29862/? D/TstMini: Observer: onDeviceReady
2022-10-29 22:06:07.754 29862-29862/? D/TstMini: SuccessCallback: onSuccess

--- AFTER 1 MINUTE DEVICE SWITCHED OFF AGAIN ---

2022-10-29 22:07:29.128 29862-29862/? D/TstMini: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2022-10-29 22:07:29.129 29862-29862/? W/TstMini: Connection lost
2022-10-29 22:07:29.130 29862-29862/? D/TstMini: Observer: onDeviceDisconnected

--- 6 MINUTES LATER SWITCHED DEVICE ON AGAIN ---

2022-10-29 22:13:12.343 29862-29862/? D/TstMini: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2022-10-29 22:13:12.344 29862-29862/? I/TstMini: Connected to A0:76:4E:71:8E:62
2022-10-29 22:13:12.344 29862-29862/? D/TstMini: wait(300)
2022-10-29 22:13:12.344 29862-29862/? D/TstMini: Observer: onDeviceConnected
2022-10-29 22:13:12.645 29862-29862/? D/TstMini: gatt.discoverServices()
2022-10-29 22:13:12.690 29862-29862/? I/TstMini: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2022-10-29 22:13:12.891 29862-29862/? I/TstMini: Services discovered
2022-10-29 22:13:12.892 29862-29862/? D/TstMini: gatt.setCharacteristicNotification(00001a00-0000-1000-8000-00805f9b34fb, true)
2022-10-29 22:13:12.895 29862-29862/? D/TstMini: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2022-10-29 22:13:12.915 29862-29862/? I/TstMini: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
[... cut ...]
2022-10-29 22:13:13.008 29862-29862/? D/TstMini: gatt.readCharacteristic(00001a01-0000-1000-8000-00805f9b34fb)
2022-10-29 22:13:13.276 29862-29862/? I/TstMini: Read Response received from 00001a01-0000-1000-8000-00805f9b34fb, value: (0x) 30-30-31-31-32-32-33-33-34-34-35-35-36-36-37-37-38-38-39-39-30-30-31-31-32-32-33-33-34-34-35-35-36-36-37-37-38-38-39-39-30-30-31-31
2022-10-29 22:13:13.278 29862-29862/? D/TstMini: FDS onDataReceived: 00112233445566778899001122334455667788990011
2022-10-29 22:13:13.278 29862-29862/? D/TstMini: Observer: onDeviceReady

Leopold84, 2022-11-04:
OK. Thank you!

closed 2022-11-04 by Leopold84