nordicsemi/Android-BLE-Library #437: no.nordicsemi.android.ble.exception.RequestFailedException

daidaijie, 2022-11-10:
We have an exception in the library using version 2.6.0-alpha03(also on 2.5.1). Is there any way to fix or catch this exception to prevent the app from crashing?

daidaijie, 2022-11-10:
`
no.nordicsemi.android.ble.ktx.RequestSuspendKt$suspendNonCancellable$2$2.onRequestFailed(RequestSuspend.kt:379)
no.nordicsemi.android.ble.Request.lambda$notifyFail$2$no-nordicsemi-android-ble-Request(Request.java:1266)
no.nordicsemi.android.ble.Request$$ExternalSyntheticLambda3.run(Unknown Source:6)
no.nordicsemi.android.ble.Request$1.post(Request.java:172)
no.nordicsemi.android.ble.Request.notifyFail(Request.java:1263)
no.nordicsemi.android.ble.BleManagerHandler$3.onDescriptorWrite(BleManagerHandler.java:2503)
android.bluetooth.BluetoothGatt$1$11.run(BluetoothGatt.java:650)
android.os.Handler.handleCallback(Handler.java:938)
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loopOnce(Looper.java:226)
android.os.Looper.loop(Looper.java:313)
android.app.ActivityThread.main(ActivityThread.java:8663)
ava.lang.reflect.Method.invoke(Native Method)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
`

daidaijie, 2022-11-10:
Device is Samsung note10+, Android 12

daidaijie, 2022-11-10:
Crash message isno.nordicsemi.android.ble.exception.RequestFailedException Request failed with status 133

philips77, 2022-11-10:
If you're using `suspend()` method you may either wrap it in `try-catch` or add a global exception handler for the `launch { ... }` block.

philips77, 2022-11-10:
As you can see here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/8653cfae7d94799a926acc2f4fd1220da1abfeb7/ble-ktx/src/main/java/no/nordicsemi/android/ble/ktx/RequestSuspend.kt#L22-L28
this method may throw some exceptions.

If that's a `ConnectRequest`, you may also use `.retry(x, delay)` method, so it will retry number of times:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/8653cfae7d94799a926acc2f4fd1220da1abfeb7/ble/src/main/java/no/nordicsemi/android/ble/ConnectRequest.java#L169-L174

closed 2022-11-19 by philips77