nordicsemi/Android-BLE-Library #438: IllegalStateException: Too many receivers

archie94, 2022-11-11:
Hello!

I am trying to connect to a LE device periodically via WorkManager. On certain devices I am getting the following crash. Is there any situation when we are leaking some receiver instance? 

Do we have to close some resources after job done to unregister the receiver which `no.nordicsemi.android.ble.BleManager` may be using?

Thanks!

```
Fatal Exception: java.lang.IllegalStateException: Too many receivers, total of 1000, registered for pid: 19004, callerPackage: a.b.c.
       at android.os.Parcel.createExceptionOrNull(Parcel.java:2434)
       at android.os.Parcel.createException(Parcel.java:2410)
       at android.os.Parcel.readException(Parcel.java:2393)
       at android.os.Parcel.readException(Parcel.java:2335)
       at android.app.IActivityManager$Stub$Proxy.registerReceiverWithFeature(IActivityManager.java:5560)
       at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1776)
       at android.app.ContextImpl.registerReceiver(ContextImpl.java:1730)
       at android.app.ContextImpl.registerReceiver(ContextImpl.java:1718)
       at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:726)
       at no.nordicsemi.android.ble.BleManager.<init>(BleManager.java:176)
       at no.nordicsemi.android.ble.livedata.ObservableBleManager.<init>(ObservableBleManager.java:24)
       at a.b.b.RBleManager.<init>(RBleManager.kt:102)
       at a.b.b.RBleManager.<init>(RBleManager.kt:92)
       at a.b.c.tina.PeriodicRDataModule.provideBleManager(PeriodicRDataModule.java:45)
       at a.b.c.tina.PeriodicRDataModule_ProvideBleManagerFactory.provideBleManager(PeriodicRDataModule_ProvideBleManagerFactory.java:69)
       at a.b.c.tina.DaggerPeriodicRDataComponent.getRBleManager(DaggerPeriodicRDataComponent.java:31)
       at a.b.c.tina.DaggerPeriodicRDataComponent.injectPeriodicRDataWorker(DaggerPeriodicRDataComponent.java:50)
       at a.b.c.tina.DaggerPeriodicRDataComponent.inject(DaggerPeriodicRDataComponent.java:44)
       at a.b.c.tina.PeriodicRDataWorker$1.invokeSuspend(PeriodicRDataWorker.kt:47)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:201)
       at android.os.Looper.loop(Looper.java:288)
       at android.app.ActivityThread.main(ActivityThread.java:7941)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1009)
```

philips77, 2022-11-19:
Are you using a single `BleManager` instance, or each time a new one?
The receivers should automatically be unregistered when the device disconnects and `close()` is called internally. Are you using `useAutoConnect = false`?

philips77, 2022-11-19:
You may try canaryleak library from Square to investigate you leaks.

archie94, 2022-11-19:
> Are you using a single `BleManager` instance, or each time a new one? The receivers should automatically be unregistered when the device disconnects and `close()` is called internally. Are you using `useAutoConnect = false`?

We do use several `BleManager` instance in different view controllers and Worker instances (as in this case). But we do call `disconnect` explicitly. 

We use `useAutoConnect(true)`

philips77, 2022-11-23:
For a single connection, like in a Worker, use `useAutoConnect(false)`. Auto connect will cause the device to reconnect on link loss automatically. Usually this is used when you have a service running, or similar, and want to maintain long connection.
Anyway, `disconnect()` should call `close()` when the device gets disconnected.

Could you share logs?

philips77, 2022-11-24:
I was interested whether you're getting a logs about closing the managers. If you override `log(...)` method, and override `getMinLogLevel()` to return `Log.DEBUG`, you should see e.g. where "close()" is called, etc.

I'm waiting for updates regarding a single manager instance.

archie94, 2022-12-16:
3 weeks into the single manager instance implementation and we have not seen this crash re-occurring. I think there laid the problem - multiple manager instance and with `useAutoConnect(true)`.

Thanks for your help. Closing this!

closed 2022-12-16 by archie94