nordicsemi/Android-BLE-Library #439: 2.6.0-alpha03 issue with Android 13

dononcharles, 2022-11-18:
Hi @philips77 
I do not know if what i've below written will help you.

I've tried to use your latest Alpha version (2.6.0-alpha03) in my BLE project but i've experienced a bad result with Android 13. With Android 12, it works.

But my problem with this Android 13 device is: i am not getting any log to put here for troubleshooting.
With this Alpha version i can successfully connect to my BLE device and send message. The issue is the BLE device reply me but i can not get any notification from the device with my Android 13 phone. This has been tested on different Google pixel Android 13 phones.

But with the official release v2.5.1, everything works as intended.

I wish i can understand the why

philips77, 2022-11-19:
Hi, we're also testing on Android 13 and have found no such issue.
Could you provide your code snippet?

philips77, 2022-11-19:
Also, to get logs, override `log(...)` method in the manager implemention.

dononcharles, 2022-11-21:
Hi @philips77 
This is the logs after i've put override log(...).

You can realize that in the v2.6.0-alpha03 logs, the reading or writing 'value' on the characteristics/Descriptors is empty.

***Version 2.6.0-alpha03 with Android 13

```kotlin
  --- logPriority:2; ---- LogMessage: Connecting...
  --- logPriority:3; ---- LogMessage: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
  connect() - device: EC:1B:BD:1B:12:A1, auto: false
  registerApp()
  registerApp() - UUID=15345000-ac88-4adb-a3ed-60543b0c5d50
  onClientRegistered() - status=0 clientIf=7
  CONNECTING TO EC:1B:BD:1B:12:A1
  onClientConnectionState() - status=0 clientIf=7 device=EC:1B:BD:1B:12:A1
 --- logPriority:3; ---- LogMessage: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
  --- logPriority:4; ---- LogMessage: Connected to EC:1B:BD:1B:12:ED
  --- logPriority:3; ---- LogMessage: wait(1600)
  onConnectionUpdated() - Device=EC:1B:BD:1B:12:ED interval=12 latency=0 timeout=400 status=0
  --- logPriority:4; ---- LogMessage: Connection parameters updated (interval: 15.0ms, latency: 0, timeout: 4000ms)
  --- logPriority:2; ---- LogMessage: Discovering services...
  --- logPriority:3; ---- LogMessage: gatt.discoverServices()
  discoverServices() - device: EC:1B:BD:1B:12:ED
  onSearchComplete() = Device=EC:1B:BD:1B:12:ED Status=0
  --- logPriority:4; ---- LogMessage: Services discovered
  --- logPriority:2; ---- LogMessage: Primary service found
  --- logPriority:3; ---- LogMessage: gatt.setCharacteristicNotification(a9da6040-0823-4995-94ec-9ce41ca28833, false)
  setCharacteristicNotification() - uuid: a9da6040-0823-4995-94ec-9ce41ca28833 enable: false
  --- logPriority:2; ---- LogMessage: Disabling notifications and indications for a9da6040-0823-4995-94ec-9ce41ca28833
  --- logPriority:3; ---- LogMessage: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x00-00)
  Controller GATT services discovered
  --- logPriority:4; ---- LogMessage: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: 
  --- logPriority:3; ---- LogMessage: gatt.setCharacteristicNotification(a73e9a10-628f-4494-a099-12efaf72258f, true)
  setCharacteristicNotification() - uuid: a73e9a10-628f-4494-a099-12efaf72258f enable: true
  --- logPriority:2; ---- LogMessage: Enabling notifications for a73e9a10-628f-4494-a099-12efaf72258f
  --- logPriority:3; ---- LogMessage: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
  --- logPriority:4; ---- LogMessage: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: 
  --- logPriority:4; ---- LogMessage: Target: EC:1B:BD:1B:12:ED is initialized
  CONNECTED TO EC:1B:BD:1B:12:ED
  --------REQUESTING FIRST COMMAND---------
  --- logPriority:2; ---- LogMessage: Writing characteristic a9da6040-0823-4995-94ec-9ce41ca28833 (WRITE COMMAND)
  --- logPriority:3; ---- LogMessage: gatt.writeCharacteristic(a9da6040-0823-4995-94ec-9ce41ca28833, value=0x4C0A040801100312111014189806220A1062F5E5, WRITE COMMAND)
  --- logPriority:4; ---- LogMessage: Data written to a9da6040-0823-4995-94ec-9ce41ca28833, value: 
  --- logPriority:2; ---- LogMessage: Writing characteristic a9da6040-0823-4995-94ec-9ce41ca28833 (WRITE COMMAND)
  --- logPriority:3; ---- LogMessage: gatt.writeCharacteristic(a9da6040-0823-4995-94ec-9ce41ca28833, value=0xD58A28C31321B3, WRITE COMMAND)
  --- logPriority:4; ---- LogMessage: Data written to a9da6040-0823-4995-94ec-9ce41ca28833, value: 
  APP SENT (0x) 4C-0A-04-08-01-10-03-12-11-10-14-18-98-06-22-0A-10-62-F5-E5-D5-8A-28-C3-13-21-B3
  Inactivity, disconnecting from the service
  disconnectFromBleController 
  --- logPriority:2; ---- LogMessage: Disconnecting...
  --- logPriority:3; ---- LogMessage: gatt.disconnect()
  cancelOpen() - device: EC:1B:BD:1B:12:ED
  onDisconnected device
  onClientConnectionState() - status=0 clientIf=7 device=EC:1B:BD:1B:12:ED
  --- logPriority:3; ---- LogMessage: [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
  --- logPriority:4; ---- LogMessage: Disconnected
  --- logPriority:3; ---- LogMessage: gatt.close()
```

***Version 2.5.1 with Android 13

```kotlin
 --- logPriority:2; ---- LogMessage: Connecting...
  --- logPriority:3; ---- LogMessage: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
  connect() - device: EC:1B:BD:1B:12:A1, auto: false
  registerApp()
  registerApp() - UUID=15345000-ac88-4adb-a3ed-60543b0c5d50
  onClientRegistered() - status=0 clientIf=7
  CONNECTING TO EC:1B:BD:1B:12:A1
[... 19 lines cut ...]
  Controller GATT services discovered
  --- logPriority:4; ---- LogMessage: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
  --- logPriority:4; ---- LogMessage: Notifications and indications disabled
  --- logPriority:3; ---- LogMessage: gatt.setCharacteristicNotification(a73e9a10-628f-4494-a099-12efaf72258f, true)
  setCharacteristicNotification() - uuid: a73e9a10-628f-4494-a099-12efaf72258f enable: true
  --- logPriority:2; ---- LogMessage: Enabling notifications for a73e9a10-628f-4494-a099-12efaf72258f
  --- logPriority:3; ---- LogMessage: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
  --- logPriority:4; ---- LogMessage: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
  --- logPriority:4; ---- LogMessage: Notifications enabled
  --- logPriority:4; ---- LogMessage: Target: EC:1B:BD:1B:12:A1 is initialized
[... cut ...] 0-3F-09-1B-42-84-75-D9-1F-58-F1-D2-DE-99-34-AC-BD-B0-4D-B2-92-05-8A-58-99-87-92
  --- logPriority:4; ---- LogMessage: Notification received from a73e9a10-628f-4494-a099-12efaf72258f, value: (0x) E6-5E-2B-84-54-8C-5C-BA-E8-3D-41-4F-B3
  APP SENT (0x) 4C-0A-04-08-01-10-03-12-43-10-14-18-A5-CC-01-22-3B-10-77-F5-E8-D5-8A-28-C5-41-CA-2C-B5-A1-DB-F9-2F-25-1B-1E-33-20-3F-09-1B-42-84-75-D9-1F-58-F1-D2-DE-99-34-AC-BD-B0-4D-B2-92-05-8A-58-99-87-92
  APP SENT (0x) E6-5E-2B-84-54-8C-5C-BA-E8-3D-41-4F-B3
```