nordicsemi/Android-BLE-Library #443: how to write the number of out-of-range bytes in the characteristic

RivayM, 2022-11-30:
Good day ! Please, tell me, I have a protocol, the start bit is 0x80 and higher, I didn’t manage to use the writeCharacteristic method to transfer this number in the data array. (because the range is -128 +127) and this number is +128. Thank you.
       I wont to write 0x80, but I write -127, of course this is related to the range, how can I expand it? I would be helped by a unsigned Byte, but it throws errors.( I use Kotlin)

------------------------------------------------------------------------------------

Добрый день! Подскажите пожалуйста, у меня протокол, стартовый бит 0x80 и выше, у меня не получилось использовать метод writeCharacteristic для передачи этого числа в массив данных. (потому что диапазон -128 +127) и это число соответственно +128. Благодарю Вас!
       Я хочу записать 0x80, но я пишу -127, конечно, это связано с диапазоном, как его расширить? Мне бы помог unsigned Byte, но он выдает ошибки. (Я использую Kotlin)

philips77, 2022-11-30:
Hi,

When you're writing byte or bytes it's up to the sender and received to agree on decoding. 
0x80 is equal to 128. If you assume that it should be read as SInt8, or `Byte`, you may do `0x80.toByte()`, this will result in -128.
Have a look at https://pl.kotl.in/rPqHzT2kC where I have set up some short playground.

closed 2022-11-30 by philips77

RivayM, 2022-12-03:
Добрый день извините , но 0x80.toByte() ( в любом случае оставляет диапазон ( -128 + 127),  а мне нужно число +128, это число необходимо в передаче используя метод  writeCharacteristic, он принимает data- тип byte, а мне необходимо, если я хочу записать в характеристику т.е передать +128, но я отправляю -127 ( в своем вопросе я ошибся Вы правы).
Как тогда можно договориться о декодировании? Если Вы предполагаете что устройство Ble, к которому я пытаюсь обратиться) я никак не могу управлять( или что -то изменять). Может возможно как то расширить возможности отправляемых данных вместо типа Byte использовать Ubyte (но наверное это приведет (где - нибудь ) к не соответствию типа.
.
Мои предположения пока складываются от того, что устройство с которым я пытаюсь связаться никак не хочет отвечать) сопряжение выполнено). Я отправляю лишь не верную команду.
необходимо:
-----------------
byteArrayOf(0x80.toByte(),0x2F,0x3F,0x21,0x0B,0x0A)
----------------
(0x80.toByte( - я ожидаю что будет 0х80  (0b000100101000) , однако кажется что в (0b000100100111)  эквиваленте Byte переводит в отрицательное число, из за чего самое первое и важно значение устройство читает не верно)

(0b000100101000) - мне необходимо ( но оно выходит за диапазон)
(0b000100100111) - 0x80.toByte()
.
Благодарю за быстрые ответы!

RivayM, 2022-12-03:
Good afternoon, sorry, but 0x80.toByte() method ( in any case, color transfer (-128 + 127), and I need the number +128, this number needs to be used in the translation writeCharacteristic, it takes a byte data type, but I need if I want to leave in the characteristic i.e. send +128, but I send -127 (I made a mistake in my question. You
are right). I can not manage (or change something). Perhaps, as an extension of the capabilities of the data being sent, instead of the Byte type, you can use Ubyte (but this use (somewhere) is possible to not use functions.
.
My discovered objects are so far discovered from the fact that the device with any objects is associated with known objects. I'm only sending the wrong command.
necessary:
byte arrays (0x80.toByte(), 0x2F, 0x3F, 0x21, 0x0B, 0x0A)
(0x80.toByte( - I expect it to be 0x80 (0b000100101000) , however, I believe that in the (0b000100100111) equivalent Byte is translated into a negative number, due to which the very first and most important value of the device does not read correctly)

(0b000100101000) - I need it (but it's out of range)
(0b000100100111) - 0x80.toByte()
.
Thanks for the quick replies!

philips77, 2022-12-07:
0x80 is equal to -127 or 128, depending on how you want to decode it.

0b000100101000 is not 128, it's 0x128, which is 296 in decimal.
0b000100100111 is not -127, it's 0x127, which is 295 in decimal.

Have a look at: https://pl.kotl.in/qTuNpkGGV

philips77, 2022-12-07:
0x80 is 0b10000000

If you extend it to Int it may become 
`0b11111111-11111111-11111111-10000000 (-127)`
or
`0b0000000-00000000-00000000-10000000 (128)`
if you use logical AND operator to truncate leading 1s.