nordicsemi/Android-BLE-Library #444: Cannot scan more than one device

filnik, 2022-12-01:
Hi! I was trying to scan multiple devices in order to connect to the nearest one. However, the scanCallback returns always the same result even if I use this scanSettings:

```
private val scanSettings: ScanSettings =
        ScanSettings.Builder()
            .setCallbackType(CALLBACK_TYPE_ALL_MATCHES)
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setMatchMode(MATCH_MODE_STICKY).build()
```

The scan call is like this:

```
scanner.startScan(getScanFilterFrom(serviceUUID), scanSettings, scanCallback)
````

where the serviceUUID it's the string uuid of the service that we are looking for.

Am I doing something wrong?

philips77, 2022-12-01:
Hello,

Are you sure you have multiple devices advertising this Service UUID in range? You can scan for them using nRF Connect for Android to verify.
Regarding your scan settings, the `CALLBACK_TYPE_ALL_MATCHES` is a default option and I was unable to find a difference in different match modes. Can you spot one?

filnik, 2022-12-01:
Yes I'm sure because when I turn off one of the devices the other will connect and work properly. Only when they are both on, only one got selected and works.

Regarding CALLBACK_TYPE_ALL_MATCHES I've added it recently. I don't see any difference removing it though :(

However probably this would work in the past since I've not changed this part of code recently, but I've just upgraded the library to the latest version (unfortunately I didn't write this part of code originally). That's why I was wondering if it might be an issue of the library itself.

philips77, 2022-12-02:
Well... You could check logs. The code snippet you gave is not even the part of this BLE library, so I have no idea what is the problem.

filnik, 2022-12-06:
I've updated the library from `2.6.0-alpha03` to `2.6.0-alpha04` and now it works correctly 🎉🎉🎉 thank you!