nordicsemi/Android-BLE-Library #448: Failing to connect after upgrading to Android 13

SarthakJ10, 2022-12-12:
Hi!
One a One plus 8T device running Android 13, it simply won't connect to the BLE peripheral
Here are the logs from `BleManager` and the app is always timing out after initiating a service discovery. 

```
12-12-2022 12:16:12.038-0500 D | BleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 2M)
12-12-2022 12:16:12.110-0500 I | BleManager: Connected to D4:D5:BE:82:CE:31
12-12-2022 12:16:12.110-0500 D | BleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
12-12-2022 12:16:12.113-0500 D | BleManager: wait(300)
12-12-2022 12:16:12.115-0500 I | BleManager: MTU changed to: 247
12-12-2022 12:16:12.415-0500 V | BleManager: Discovering services...
12-12-2022 12:16:12.416-0500 D | BleManager: gatt.discoverServices()
12-12-2022 12:16:12.776-0500 I | BleManager: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
12-12-2022 12:23:07.385-0500 D | BleManager: [Callback] Connection state changed with status: 8 and new state: 0 (DISCONNECTED)
12-12-2022 12:23:07.388-0500 W | BleManager: Error: (0x8): GATT CONN TIMEOUT
```

Here is the `connect` code:

```kotlin
fun connectDevice(bluetoothDevice: BluetoothDevice) {
    val request = connect(bluetoothDevice)
        .useAutoConnect(false)         
        .usePreferredPhy(no.nordicsemi.android.ble.PhyRequest.PHY_LE_2M_MASK)
        .retry(CONNECTION_RETRIES, CONNECTION_RETRIES_DELAY_MS)
        .enqueue()
}
```

The device was able to connect successfully while it was running Android 12 but as soon as you update it to Android 13, it fails to connect. 

Details:
Device: One plus 8T 
OS: Android 13 (KB2005_11_F.13)
Library version: v2.5.1

archie94, 2022-12-16:
Happening for us too on some user devices. 

May be related: https://issuetracker.google.com/issues/242755161?pli=1

SarthakJ10, 2022-12-16:
Here are the sniffer logs. I have multiple attempts to try and connect to the BLE peripheral and the first one is where bonding happens. 
[withBonding.pcapng.zip](https://github.com/NordicSemiconductor/Android-BLE-Library/files/10249460/withBonding.pcapng.zip)

archie94, 2022-12-20:
This is being reported by more of our users!

To add more context we manufacture custom wearable, PCB of which are based on Nordic designs. On Android 13 certain devices are failing to connect with these wearables. The logs of these are mainly of two types, one where there is an instant disconnect after connection (pretty much similar to the one @SarthakJ10 has posted) the other type being services not being discovered after connection.

We initially thought this was some bond management issue happening for users who are bonded while on Android 12 and after the OS update to Android 13 the connection was failing. However this is not the case. For a user on Android 13 and trying to connect to our peripheral for the first time this happens as well. Even the [blinky sample](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fble_sdk_app_blinky.html) project fails to connect with the phone. 

On the same device we started our test while to Android 11 -> Android 12-> Android 13. It works as expected on Android 11 and Android 12.

Tested on  
Device: One plus 8T (Model KB2001)
OS: Android 13 (KB2001_11_F.13)
Library version: v2.5.1

Please let me know if you would need any more specific central or peripheral side logs.

philips77, 2022-12-20:
Did you try removing preferred PHY?

philips77, 2022-12-20:
Also, did you try other phones with Android 13?
I'll try to acquire One Plus 8T for testing.
Thank you for reporting the issue.

SarthakJ10, 2022-12-20:
> Did you try removing preferred PHY?

@philips77 yes. I've tried with removing preferred PHY and also experimented with other PHYs but that did not help.

SarthakJ10, 2022-12-20:
> Also, did you try other phones with Android 13?
> I'll try to acquire One Plus 8T for testing.
> Thank you for reporting the issue.

@philips77 Thank you! Yes I did try on Pixel 7 running Android 13 and I have no issues on that phone.

archie94, 2022-12-20:
> Also, did you try other phones with Android 13? I'll try to acquire One Plus 8T for testing. Thank you for reporting the issue.

Yes. We tried on Pixel 7 Pro, Pixel 6 pro and a few other phones on android 13 but they were fine.

These are the user devices on which our app seemed buggy

- KB2001-  OP 8T (*3)
- SM-S901E - [Samsung Galaxy S22 5G](https://www.gsmarena.com/samsung_galaxy_s22_5g-11253.php)
- NE2211 - one plus 10 Pro 5G (*2)
- SM-S908E [Samsung Galaxy S22 Ultra (SM-S908E/DS 256GB) All specs](https://www.phonemore.com/specs/samsung/galaxy-s22-ultra/sm-s908eds-256gb/)

SarthakJ10, 2022-12-22:
I am not sure if the Bose Music App uses Nordic library under the hood but I was able to connect a pair of Bose headphones using their [app](https://play.google.com/store/apps/details?id=com.bose.bosemusic&hl=en_US&gl=US) on One plus 8T running Android 13
I have attached the sniffer log below

[boseHeadphonesPublic.pcapng.zip](https://github.com/NordicSemiconductor/Android-BLE-Library/files/10288416/boseHeadphonesPublic.pcapng.zip)

archie94, 2022-12-23:
> I am not sure if the Bose Music App uses Nordic library under the hood but I was able to connect a pair of Bose headphones using their [app](https://play.google.com/store/apps/details?id=com.bose.bosemusic&hl=en_US&gl=US) on One plus 8T running Android 13 I have attached the sniffer log below

I thought headphones use Bluetooth classic protocol and not LE.

archie94, 2022-12-30:
> We initially thought this was some bond management issue happening for users who are bonded while on Android 12 and after the OS update to Android 13 the connection was failing. However this is not [...]

So we tried a sample [project](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/bluetooth/peripheral_lbs/README.html) (based on nrf connect sdk instead of nrf softdevice) and it works as expected on the One Plus 8T device. Connection works and services are getting discovered. Although I am not sure what this means and where the solution would lies.

SarthakJ10, 2023-01-05:
@philips77 any updates here?

archie94, 2023-01-05:
@SarthakJ10 

We are in control of the peripheral and trying to build our project based on nrf sdk v16.0.0. 

Sample projects running on nrf sdk v16.0.0 connect with One Plus 8T. The same is not the case with nrf sdk v17.1.0.560.  
If you are in control of the firmware running can you also try some experiments?

SarthakJ10, 2023-01-05:
> @SarthakJ10
>
> We are in control of the peripheral and trying to build our project based on nrf sdk v16.0.0.
>
> Sample projects running on nrf sdk v16.0.0 connect with One Plus 8T. The same is not the [...]

@archie94 I checked in with the FW Engineering team, we use v17.0.2 and unfortunately downgrading a major release is not a viable option for us.

SarthakJ10, 2023-01-09:
@archie94 did you try to contact One plus support team too? Curious if you had any leads there? I did and I am still yet to receive a concrete reply from them.

archie94, 2023-01-10:
Hey @SarthakJ10 

No. Can you point me to their developer support forum?

32penkin, 2023-02-15:
Hi guys,
Looks like the same issue with:
- 2109119DG - Xiaomi 11 Lite 5G NE

At the same time works as expected with the:
- SM-G990B/DS - Samsung Galaxy S21 FE 5G 

I use android-ble-library v.2.2.4.
@archie94 , @SarthakJ10 , do you guys have any updates on this?

philips77, 2023-03-07:
This seems to be related to phones sending Read Multiple request in service discovery, and ignoring "Not supported" response. 
[image]

Do you have logs when using SDK 16 and 17? What is the difference there?

archie94, 2023-03-13:
hi @philips77 

I have emailed you the hex on aleksander.nowakowski@nordicsemi.no
Subject: Failing to connect after upgrading to Android 13 Github #448

archie94, 2023-03-15:
Hi @philips77 

Just checking if you have received the HEX.
Also if there are any other info I can help you with please let me know.

We run the HEX on 54840 DK.

archie94, 2023-03-16:
We are seeing that on certain devices after upgrade to android 13 it is failing with

```
Error: (0x3b): UNKNOWN (59)
```

Devices that we saw this issue:

- Asus Zenphone 9 - ASUS_AI2202
- Oneplus 11 5g - CPH2451
- Oneplus 9 5g - LE2111

philips77, 2023-03-21:
We managed to reproduce the issue on our side with OnePlus 8T and are investigating.
Neither nRF5 SDk nor NCS support Read Multiple request, but the central should not continue after getting Not Supported error.

archie94, 2023-03-30:
@SarthakJ10 did you have any luck in circumventing this issue?

SarthakJ10, 2023-03-30:
> @SarthakJ10 did you have any luck in circumventing this issue?

@archie94 Nope. No luck so far. Although Sony released an OS update and that resolved issues just for the Sony Xperia phone. The issue still persists for One plus, Asus and even Nothing phone now. 
We are also exploring a potential hypothesis where if you remove the Bond Management Service on the BLE peripheral, it connects successfully

pdapnz, 2023-07-06:
Hi! We are running into very similar issue with OnePlus phones using BLE connеction with Android OS and out devices, after they upgrade to lastest OxygenOs 13.1. But we use another hardware in our devices - based on Texas Instruments. Is any news about this problem? May be any workaround?

Maybe anyone tried to write a report to the developers of OxygenOs, so that they can see from their side what could be the problem?

Fabbros, 2024-04-03:
@archie94 @SarthakJ10 Did you have any luck in circumventing this issue?

I'm having similar issues with a Samsung S21 with Android 14.

archie94, 2024-04-08:
> @archie94 @SarthakJ10 Did you have any luck in fixing this issue?
> 
> I'm having similar issues with a Samsung S21 and Android 14.

Our firmware team was able to resolve the issue. I do not have much idea on which parameters they changed but no change was required from application / library side.

SarthakJ10, 2024-04-08:
> @archie94 @SarthakJ10 Did you have any luck in fixing this issue?
> 
> I'm having similar issues with a Samsung S21 and Android 14.

We were not able to fix it with Android 13. But Sony and other OEM manufacturers released OS level fix for Android 13 which fixed the issues for some of our customers.