nordicsemi/Android-BLE-Library #452: The request is lost if the device is not ready

onatakduman, 2022-12-29:
Hi Ble-Library developers, the library is great I am grateful, thanks a lot.

I have an issue: when I try to add a request to queue, the request lost if device is not in `isReady ` state so the command is lost. I found a solution for that but I am not proud of it. Because it might be cause a performance issue or memory leak. 

I couldn't figure out what I did wrong. How can I solve this situation? Can you help me? 

`ViewModel` operations
```kotlin
/*find and connect to device*/
val device = bluetoothAdapter.getRemoteDevice(device.deviceAddress)
bleManager.connect(device)
    .retry(3, 100)
    .enqueue()

/* ...
some operations
...*/

/*this is the solution for guarantee the request delivery*/
if (bleManager.isConnected) {
        if (bleManager.isReady) {
            bleManager
                .rename("device-name") /*custom rename write request*/
                .enqueue()
        } else {
            viewModelScope.launch(Dispatchers.IO) {
                while (true) {
                    if (bleManager.isReady) {
                        bleManager
                            .rename("device-name") /*custom rename write request*/
                            .enqueue()
                        cancel()
                    }
                    delay(100L)
                }
            }
        }
}

```

`MyBleManager.kt`
```kotlin
class MyBleManager(
    context: Context,
    private val scope: CoroutineScope
) : BleManager(context) {

/* ... gatt operations ...*/

fun rename(name: String): WriteRequest {
        val text = "$name"
        val writeType = BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE
        val request: WriteRequest =
            writeCharacteristic(characteristicRename, text.toByteArray() + 0, writeType)
        /* I tried suspend version but result is same*/
        return request.with(dataCallback)
}

/* ... other functions etc. ...*/

}
```
Thanks and best regards :)

CyDragon80, 2023-01-05:
I'm still somewhat new to using this library myself, but I think I have some suggestions. When you call "enqueue" you are adding to a list of async operations that are executed FIFO in a handler thread, so execution will continue after the call without blocking for the operation to be completed. I think you can attach callbacks to requests with "done" or "then" to be performed after the operation is complete. Think they show some of the request callbacks in the README https://github.com/NordicSemiconductor/Android-BLE-Library/blob/main/README.md?plain=1#L252
Also if you just want to do stuff after connection is established, you can probably override initialize as well, but maybe the "done" callback would make more sense in this case? https://github.com/NordicSemiconductor/Android-BLE-Library/blob/main/examples/ble-gatt-client/src/main/java/no/nordicsemi/android/ble/ble_gatt_client/GattService.kt#L228

onatakduman, 2023-01-06:
@CyDragon80 Thank you for your help and investigation about it, You are actually right about that, if I put the sending commands in connection requests `done` callback, the manager is getting `ready` to send it, but handled it in `else` case of the function "`if (bleManager.isConnected)`" I forget to add in the issue.

I'm trying to make a **connection fast** and send a command so connection is handled with **proximity sensor** if the user gets closer to the device (not a phone or tablet) connection starts, if the proximity sensor is not interrupted, I use display touches for make a connection. After the interaction, if the ble manager is connected, I can send the command, so I am trying to do it that way.

In that device connection takes **2**-**8** **seconds**, depending on something I didn't figure out (maybe the Bluetooth chip is not good enough). 

I can't write the whole code because of "company issues". 

```kotlin
/*this is the solution for guarantee the request delivery*/
if (bleManager.isConnected) {
        if (bleManager.isReady) {
            bleManager
                .rename("device-name") /*custom rename write request*/
                .enqueue()
        } else {
            viewModelScope.launch(Dispatchers.IO) {
                while (true) {
                    if (bleManager.isReady) {
                        bleManager
                            .rename("device-name") /*custom rename write request*/
                            .enqueue()
                        cancel()
                    }
                    delay(100L)
                }
            }
        }
} else {
        /*else case not showing in the issue*/
        bleManager.connect(remoteDevice)
            .done {
                if (bleManager.isReady) {
                    bleManager
                        .rename("device-name") /*custom rename write request*/
                        .enqueue()
                }
            }
            .retry(4, 250)
            .enqueue()
}
```

corentin-c, 2023-01-09:
Hello,

This might not be the best solution for what you want to do, but you could improve it by overriding onDeviceReady in your BleManagerGattCallback and calling your rename request only here, rather than doing 

> while (true) {
>                     if (bleManager.isReady) {

bleManager.isReady is set to true only when onDeviceReady is called so this should do the same if I understand it correctly

philips77, 2023-01-09:
Hi, 
Requests enqueued after the connect request was added should not be ignored. They should be executed after the initialization queue is complete.

```kotlin
connect(remoteDevice)
   .retry(4, 250)
   .enqueue()
writeCharacteristic(characteristicRename, text.toByteArray() + 0, writeType)
   .enqueue()
```

This enqueues 2 requests: `ConnectRequest` and `WriteRequest`. When you enqueue the 2nd while 1 is still in progress, it will be added for later execution. Look here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/cdd6beb7f5ba94d9d97755cfb56d24da7c3d229b/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1452-L1460
 
One thing I've noticed it perhaps it would be better to use a flag whether the tasks are added from `initialize()` or not. Currently it's based on the presence of `initQueue`, but this may exist when the service have been discovered until all the tasks are completed. So it's possible to add an init task from outside of `initialize()`. Anyway, the task should be properly executed.

philips77, 2023-01-10:
Have a look at https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/c3af703d8cc4ab0541dcdcf3eda84353a3cfb339/mcumgr-ble/src/main/java/io/runtime/mcumgr/ble/McuMgrBleTransport.java#L338-L362
In this project we're ensuring the device is connected before each write:
```java
connect(device)
  .done(device -> {
     // send data here
  }
  .enqueue()
```
If the device is already connected, `done` callback is called immediately.

closed 2023-01-10 by philips77

philips77, 2023-01-10:
> One thing I've noticed it perhaps it would be better to use a flag whether the tasks are added from initialize() or not. Currently it's based on the presence of initQueue, but this may exist when [...]

This has been addressed in #456.

onatakduman, 2023-01-10:
> Have a look at [...]

Hi, @philips77 thanks for the explanation directly I'll change my code as follows to your directions. :) Have a nice day