nordicsemi/Android-BLE-Library #453: How to merge data more than 20 bytes.

karthick1611, 2023-01-10:
Hi,

I implemented your library, When I try to test it with Samsung Galaxy S22 series, Bluetooth 5.2 and Android OS version 13, I can't able to set MTU as 512. When I set MTU as 512 nothing working so I removed the request MTU, now my default MTU was like 23. 

Now I'm writing characteristics, writing was working fine but I can able to read only 20 bytes of data.

How to read remaining data?.

I used Split() while writing Characteristics. 

`2023-01-09 16:21:34.866 26215-26215 MyBleManager            com.library.blenew                   V  Connecting...
2023-01-09 16:21:34.866 26215-26215 MyBleManager            com.library.blenew                   D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2023-01-09 16:21:34.866 26215-26215 BluetoothAdapter        com.library.blenew                   I  STATE_ON
2023-01-09 16:21:34.868 26215-26215 BluetoothGatt           com.library.blenew                   D  connect() - device: 500291_6, auto: false
2023-01-09 16:21:34.868 26215-26215 BluetoothGatt           com.library.blenew                   D  registerApp()
2023-01-09 16:21:34.868 26215-26215 BluetoothGatt           com.library.blenew                   D  registerApp() - UUID=b2226d20-db3d-4124-b290-ccd612b32a40
2023-01-09 16:21:34.871 26215-26230 BluetoothGatt           com.library.blenew                   D  onClientRegistered() - status=0 clientIf=6
2023-01-09 16:21:35.107 26215-26266 BluetoothGatt           com.library.blenew                   D  onClientConnectionState() - status=0 clientIf=6 device=500291_6
2023-01-09 16:21:35.115 26215-26215 MyBleManager            com.library.blenew                   D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2023-01-09 16:21:35.116 26215-26215 MyBleManager            com.library.blenew                   I  Connected to 50:02:91:81:21:E6
2023-01-09 16:21:35.120 26215-26215 MyBleManager            com.library.blenew                   D  wait(300)
2023-01-09 16:21:35.420 26215-26215 MyBleManager            com.library.blenew                   V  Discovering services...
2023-01-09 16:21:35.420 26215-26215 MyBleManager            com.library.blenew                   D  gatt.discoverServices()
2023-01-09 16:21:35.420 26215-26215 BluetoothGatt           com.library.blenew                   D  discoverServices() - device: 500291_6
2023-01-09 16:21:35.854 26215-26266 BluetoothGatt           com.library.blenew                   D  onConnectionUpdated() - Device=500291_6 interval=6 latency=0 timeout=500 status=0
2023-01-09 16:21:35.856 26215-26215 MyBleManager            com.library.blenew                   I  Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2023-01-09 16:21:36.117 26215-26272 BluetoothGatt           com.library.blenew                   D  onSearchComplete() = Device=500291_6 Status=0
2023-01-09 16:21:36.118 26215-26215 MyBleManager            com.library.blenew                   I  Services discovered
2023-01-09 16:21:36.118 26215-26215 Read                    com.library.blenew                   E  6e400003-b5a3-f393-e0a9-e50e24dcca9e
2023-01-09 16:21:36.118 26215-26215 Write                   com.library.blenew                   E  6e400002-b5a3-f393-e0a9-e50e24dcca9e
2023-01-09 16:21:36.118 26215-26215 MyBleManager            com.library.blenew                   V  Primary service found
2023-01-09 16:21:36.123 26215-26215 MyBleManager            com.library.blenew                   D  gatt.setCharacteristicNotification(6e400003-b5a3-f393-e0a9-e50e24dcca9e, true)
2023-01-09 16:21:36.123 26215-26215 BluetoothGatt           com.library.blenew                   D  setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2023-01-09 16:21:36.124 26215-26215 MyBleManager            com.library.blenew                   V  Enabling notifications for 6e400003-b5a3-f393-e0a9-e50e24dcca9e
2023-01-09 16:21:36.124 26215-26215 MyBleManager            com.library.blenew                   D  gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2023-01-09 16:21:36.141 26215-26215 MyBleManager            com.library.blenew                   I  Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
2023-01-09 16:21:36.142 26215-26215 MyBleManager            com.library.blenew                   V  Requesting new MTU...
2023-01-09 16:21:36.142 26215-26215 MyBleManager            com.library.blenew                   D  gatt.requestMtu(512)
2023-01-09 16:21:36.142 26215-26215 BluetoothGatt           com.library.blenew                   D  configureMTU() - device: 500291_6 mtu: 512
2023-01-09 16:21:36.146 26215-26215 MyBleManager            com.library.blenew                   I  Target initialized
2023-01-09 16:21:36.161 26215-26266 BluetoothGatt           com.library.blenew                   D  onConfigureMTU() - Device=500291_6 mtu=512 status=0
2023-01-09 16:21:36.162 26215-26215 MyBleManager            com.library.blenew                   I  MTU changed to: 512
2023-01-09 16:21:36.163 26215-26215 MyBleManager            com.library.blenew                   V  Writing descriptor 00002902-0000-1000-8000-00805f9b34fb
2023-01-09 16:21:36.163 26215-26215 MyBleManager            com.library.blenew                   D  gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x0100)
2023-01-09 16:21:36.176 26215-26215 MyBleManager            com.library.blenew                   I  Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
2023-01-09 16:21:36.176 26215-26215 MyBleManager            com.library.blenew                   V  Writing characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e (WRITE REQUEST)
2023-01-09 16:21:36.176 26215-26215 MyBleManager            com.library.blenew                   D  gatt.writeCharacteristic(6e400002-b5a3-f393-e0a9-e50e24dcca9e, value=0x7375204D616B654C6F74734F6640730A, WRITE REQUEST)
2023-01-09 16:21:36.194 26215-26215 MyBleManager            com.library.blenew                   I  Data written to 6e400002-b5a3-f393-e0a9-e50e24dcca9e
2023-01-09 16:21:36.195 26215-26215 MyBleManager            com.library.blenew                   V  Writing characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e (WRITE REQUEST)
2023-01-09 16:21:36.195 26215-26215 MyBleManager            com.library.blenew                   D  gatt.writeCharacteristic(6e400002-b5a3-f393-e0a9-e50e24dcca9e, value=0x677665720A, WRITE REQUEST)
2023-01-09 16:21:36.200 26215-26215 Data Sent               com.library.blenew                   E  su MakeLotsOf@s
2023-01-09 16:21:36.201 26215-26215 MyBleManager            com.library.blenew                   I  Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 23-0D
2023-01-09 16:21:36.201 26215-26215 MyBleManager            com.library.blenew                   I  Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 7B-0D
2023-01-09 16:21:36.201 26215-26215 MyBleManager            com.library.blenew                   I  Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 20-20-22-4C-6F-67-69-6E-22-3A-22-43-6F-72-72-65-63-74-22-0D
2023-01-09 16:21:36.206 26215-26338 Response                com.library.blenew                   E  #
2023-01-09 16:21:36.207 26215-26338 Response                com.library.blenew                   E  {
2023-01-09 16:21:36.207 26215-26340 Response                com.library.blenew                   E    "Login":"Correct"
2023-01-09 16:21:36.209 26215-26215 MyBleManager            com.library.blenew                   I  Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 7D-0D
2023-01-09 16:21:36.209 26215-26215 MyBleManager            com.library.blenew                   I  Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 23-0D
2023-01-09 16:21:36.210 26215-26338 Response                com.library.blenew                   E  }
2023-01-09 16:21:36.210 26215-26338 Response                com.library.blenew                   E  #
2023-01-09 16:21:36.251 26215-26272 BluetoothGatt           com.library.blenew                   D  onConnectionUpdated() - Device=500291_6 interval=36 latency=0 timeout=500 status=0
2023-01-09 16:21:36.251 26215-26215 MyBleManager            com.library.blenew                   I  Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
[... cut ...]
2023-01-09 16:21:41.591 26215-26215 MyBleManager            com.library.blenew                   D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2023-01-09 16:21:41.591 26215-26215 MyBleManager            com.library.blenew                   I  Disconnected
2023-01-09 16:21:41.604 26215-26215 MyBleManager            com.library.blenew                   D  gatt.close()
2023-01-09 16:21:41.604 26215-26215 BluetoothGatt           com.library.blenew                   D  close()`

When I set request MTU as 512, Writing getting failed and disconnected.

karthick1611, 2023-01-10:
fun writeData(data : String) {
        writeCharacteristic(
            fluxCapacitorControlPointWrite,
            "$data\n".toByteArray(),
            BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT
        )
//            .split(DataSplitter().chunk("$data\n".toByteArray(), 0, 20))
//            .split(DataSplitter { message, index, maxLength -> "$data\n".toByteArray()})
            .split()
            .with { device, data -> Log.e("Data Sent", String(data.value!!, Charsets.UTF_8)) }
//            .split(DataSplitter { message, index, _ -> byteArrayOf(message[index]) })
            .enqueue()
    }

philips77, 2023-01-10:
Hi,
If the remote device accepts MTU = 512 then sending longer packets should work. Can you say what is not working exactly? Are you sure there are no issues on the firmware side? If you're using Nordic devices, please consult on [DevZone](https://devzone.nordicsemi.com/).

However, whether MTU is 23 (max 20 bytes), or 512 (max 509 bytes of payload), if your actual payload is longer than that you need to split it on one side to fit packets and merge on the other side.

## Splitting and merging

How to split and merge depend on the device specification. The received needs to know when to stop merging and return the reassembled value. The total size may be known beforehand, sent in the first packet, marked using a flag in each packet, etc. The sender implementation must match the received.

#### JSON

For example, let's say you want to send a JSON file. You don't have to send the length, as the receiver may try to parse the received data each time it gets a new packet. For that purpose use `JsonMerger` on the receiver (if you're using Android on server side), or implement similar thing in the firmware.

#### Length
If the data is not a JSON file, or you don't want to spend time/battery to parse it after each packet, you may decide to send the total length in bytes before the data.
The first packet may contain 2 bytes of "total length in UINT16 in Little Endian" followed by some number of packets with given number of bytes.
The receiver would receive the length, set up a buffer and collect packets until the desired length is obtained.

#### Flags
You may also append a flag to each packet sent. Let's say the first byte in each indicates "the last packet" or "more will come". The received buffers incoming packets (skipping the first byte) until the last packet is received.

#### Known size
If the size of the packet is known you may completely skip sending length, as the received knows the size.

philips77, 2023-01-10:
Have a look s `PacketSlitter` and `PacketMerger` here: https://github.com/NordicSemiconductor/Android-BLE-Library/tree/main/examples/trivia/src/main/java/no/nordicsemi/android/ble/trivia/spec

philips77, 2023-01-10:
When using `.split()` you'll receive `(MTU - 4 + n) / (MTU-3)` notifications on the receiver side, each containing 20 or less bytes. You need to merge them to get the full data.

karthick1611, 2023-01-10:
Ok. Is there any code snippet is there to merge the data?. I didn't find any sample code with it...

philips77, 2023-01-10:
I already gave you the classes here: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/453#issuecomment-1377172734. The splitter and merger are using the "total length" approach. The first packet sent is prefixed with 2 byte length field.

karthick1611, 2023-01-10:
> Also, seems like you're sending each line of the JSON separately for some reason. You may send it all together, better without line breaks to save some bytes. Only then you know the total size and can parse the JSON.

Json is response. "setconf serialNumber" this is command I'm passing to.