nordicsemi/Android-BLE-Library #455: Device connected but not ready

talfsaverone, 2023-01-10:
Hi,

In some cases, I receive `onDeviceConnected`  callback but `onDeviceReady` is not triggered.

Also, the `isRequiredServiceSupported` method does not call.

Turning the device Bluetooth on and off is a workaround for this, but that's not a valid solution.

philips77, 2023-01-10:
Looks like something fails with service discovery. Do you get disconnection or an error after some time? Are you using `.retry(times, delay)` in `ConnectRequest`?

talfsaverone, 2023-01-10:
We are not setting retry.
But we are not getting the device disconnected or any failed callback.
The devices remain connected on the Bluetooth page.

What is the recommended retry settings?

philips77, 2023-01-10:
I'm usually using `retry(3, 100)`, so 3 times with delay 100 ms in between. Seems to be fairly stable...

philips77, 2023-01-10:
Here's 3, 300: https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/8352e92ce2dd12af5a05a60643a64187473adbc0/blinky/ble/src/main/java/no/nordicsemi/android/blinky/ble/BlinkyManager.kt#L57

closed 2023-01-11 by talfsaverone