nordicsemi/Android-BLE-Library #460: Question about invalid response package error

adi-pilav-sn, 2023-01-17:
Hi there!

I am experiencing an issue with Samsung S21 Ultra 5G phone running on Android 13 described [here](https://issuetracker.google.com/u/1/issues/265343637).

While I am waiting for the responses on the reported issue, I am trying out different MTU sizes as the reported issue was reproducible with MTU size 513.
When I am using the lowest value for the MTU - 23 everything works fine, however, if I want to use some other value like 256 I receive this error:
```
11:02:26.127  E  [ERROR:gatt_cl.cc(1244)] gatt_client_handle_server_rsp: invalid response pkt size: 513, PDU size: 256
11:02:26.127  I  [INFO:bta_gattc_main.cc(314)] bta_gattc_sm_execute: State 0x02, Event 0x1f0f
11:02:26.145  E  onCharacteristicRead error 133
11:02:26.148  E  Nordic|Error (0x85): GATT ERROR
```
I looked into the `btsnoop_hci.log` I don't see any packages larger than the MTU size:
[image]

I request MTU in two places:
- during the connection initialization:
```
manager.requestMtu(ANDROID_MTU_MAX_SIZE)
            .with { _, mtu -> logD { "handleInitialization: new mtu $mtu" } }
            .coAwait()
```
- and in the `onMtuChanged` callback:
```
override fun onMtuChanged(gatt: BluetoothGatt, mtu: Int) {
                logD { "Mtu was changed to $mtu" }

                if (mtu > ANDROID_MTU_MAX_SIZE) {
                    logD { "Requesting MTU to be $ANDROID_MTU_MAX_SIZE" }
                    manager.requestMtu(ANDROID_MTU_MAX_SIZE).enqueue()
                }
            }
```
and I log MTU on each request, so I know that MTU was not changed.

Did anybody experience this issue or [the previously mentioned one](https://issuetracker.google.com/u/1/issues/265343637)?

If someone can give me a hint or any advice it would be much appreciated!

philips77, 2023-01-18:
Can it be related to Data Length Extension? Did you try MTU = 247, the maximum which can fit into a single L2CAP packet?

adi-pilav-sn, 2023-01-19:
@philips77 thanks for the hint!
I tried out MTU 247 and other values and they all failed on Samsung S21 Ultra 5G running Android 13:
- MTU 247:
```
12:32:15.483  I  [INFO:bta_gattc_main.cc(314)] bta_gattc_sm_execute: State 0x02, Event 0x1f04
12:32:15.625  E  [ERROR:gatt_cl.cc(1244)] gatt_client_handle_server_rsp: invalid response pkt size: 513, PDU size: 247
12:32:15.626  I  [INFO:bta_gattc_main.cc(314)] bta_gattc_sm_execute: State 0x02, Event 0x1f0f
12:32:15.628  E  onCharacteristicRead error 133
12:32:15.629  E  Nordic|Error (0x85): GATT ERROR
```
- MTU 240:
```
[ERROR:gatt_cl.cc(1244)] gatt_client_handle_server_rsp: invalid response pkt size: 513, PDU size: 240
12:38:59.837  I  [INFO:bta_gattc_main.cc(314)] bta_gattc_sm_execute: State 0x02, Event 0x1f0f
12:38:59.844  E  onCharacteristicRead error 133
12:38:59.845  E  Nordic|Error (0x85): GATT ERROR
```
- MTU 50:
```
13:36:47.166  E  [ERROR:gatt_cl.cc(1244)] gatt_client_handle_server_rsp: invalid response pkt size: 513, PDU size: 50
13:36:47.167  I  [INFO:bta_gattc_main.cc(314)] bta_gattc_sm_execute: State 0x02, Event 0x1f0f
13:36:47.170  E  onCharacteristicRead error 133
13:36:47.171  E  Nordic|Error (0x85): GATT ERROR
```
Until I tried using max value 517, and it worked. 

The reason why we used 513 before, is that we run into issues on Pixel 6 and Pixel 6 pro running Android 13 where the write buffer seems to be limited to 513 bytes:
```
11:42:15.157  D  writeCharacteristic() 
11:42:15.159  A  FORTIFY: memcpy: prevented 514-byte write into 513-byte buffer
11:42:15.160  A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 10885 (bt_main_thread), pid 10787 (droid.bluetooth)
11:42:15.299  I  obtaining output fd from tombstoned, type: kDebuggerdTombstoneProto
11:42:15.305  I  received crash request for pid 10885
11:42:15.306  I  performing dump of process 10787 (target tid = 10885)
```
So we limited the MTU to 513 size.

Thanks once again for the help! I will close this question now.

closed 2023-01-19 by adi-pilav-sn