nordicsemi/Android-BLE-Library #463: Issue with reading on Android 13 Samsung devices

peropavlovic, 2023-01-24:
Hi Android-Ble-Library developers,

we have issue with readCharacteristic request on Samsung devices using Android 13. Issue occurs only when we trying to read data larger then 512. We tried changing MTU sizes but nothing solve this issue. When we try to call readCharacteristic it fails with Fail reason **-100** and BLE Connection get disconnected. 

Can you help us solve this issue as it's really important as Users with those devices cannot use some of fundamental features of our app.

Thanks!

adi-pilav-sn, 2023-01-25:
@peropavlovic Have you tried requesting MTU of size 517? I had a similar issue https://issuetracker.google.com/issues/265343637 and it worked for me.

peropavlovic, 2023-01-25:
@adi-pilav-sn thanks for advice! I have tried and I'm not getting disconnected anymore but now it fails with reason 133. I checked out Device and everything looks good there.

On device scenario with MTU lower then 517 was that all chunk was sent, and on last one readCharacteristic fails and  BLE get disconnected.

Now when I tried with MTU 517, only one chunk sent and readCharacteristic fails with reason 133.

Any advice what should I do next?

adi-pilav-sn, 2023-01-25:
Can you try to paste the Android system logs here when 133 error happens? Note: if you are using Android Studio remove any tags (including package name) from the logcat and you will see the Android system logs that may give some more info about what went wrong.

philips77, 2023-01-26:
1. How long is your data that you want to send?
2. Are you using *write with response*, which would use Long Read Procedure, or *write without response*?

If you're using *write without response* you need to split the data into MTU-3 byte long packets on your own. If the size is not known on the receiver side, you need to send the length as well. Each packet can be up to MTU-3 bytes long.

If you're using *write with response* and the data are longer than MTU, Android will automatically use Long Read Procedure. Data will be split automatically if MTU is lower, but again, the longest packet can be 500 bytes, I think (or around that number). For longer packets I recommend *write without response*, as they are faster.

In the BLE library we have `DataSplitter` and `DataMerger` interfaces, which you may use.
Here you may find some useful implementations: https://github.com/NordicSemiconductor/Android-BLE-Library/tree/testing-features/test/src/main/java/no/nordicsemi/andorid/ble/test/spec
If you're connecting with device, not Android phone, you need to implement matching spitter on your own in the fw.

philips77, 2023-01-26:
For example,. the [HeaderBasedPacketMerger](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/testing-features/test/src/main/java/no/nordicsemi/andorid/ble/test/spec/HeaderBasedPacketMerger.kt) and splitter add UINT16 length field for every message. When receiving, the other device knowns how long packet to expect, can set up a buffer and collect requests until the fill buffer is full. Splitter adds the length field, merger reads it and collects.

We also have an example of flag based splitter-merger pair, where each packet contains an information: full, start, continuation, end, and based on that the merger knows when the data are complete.

If the length of the packet is known to the merger you may just use `.split()` and the lib will split it into MTU-3-byte long packets automatically. Then collect packets on the receiver side until you get required number of bytes.

philips77, 2023-01-26:
If you have some specification to follow, check how that is specified there.

peropavlovic, 2023-01-26:
Hi @philips77, thanks for the info! 

Our problem is reading data with readCharacteristic, not writing. Data they send to us can be different size and the one we have problem with is when that data is larger then 515 bytes (Usually its around 1000 bytes). And in that scenario, while reading data our readCharacteristic is failing.

peropavlovic, 2023-01-26:
@philips77 Just to explain a little bit better our flow.

We have gatt server written in GO.
Logic is that on server side they send us Data in chunks of MTU size. Let's say its 515 bytes. So on client (App) size, we read first chunk and ask for another. 

Issue is that, when trying to read first chunk with readCharacteristic() it fails without reading it (We can see on server logs that sending is good) and usually disconnects from BLE connection. 

This issue was working good on all device and Android versions, except Samsung devices with Android 13.

philips77, 2023-01-26:
I think I may know the issue then. As far as I remember, starting from Android 13 the OS is now correctly handling too long values. For Long Read or Long Write there's a limit for length is 512 bytes:

> The maximum length of an attribute value shall be 512 octets.

Core Bluetooth 5.2, Vol 3, Part F, 3.2.9.

Before it was "somehow working", but since 13 the limit is checked.
You need to split the value into smaller chunks and indicate somehow when the value is fully received. 

Btw, in the BLE library you may set a merger to `readCharacteristic` as well. It will perform number of reads until the merger returns *true*. But you need to modify the server side to match the behavior.

peropavlovic, 2023-02-02:
Hi @philips77 I still have this issue and trying to figure it out how to solve it. My users keep pushing me as they cannot use this major feature because of this issue.

So we found out it occurs on Samsung devices with Android 13. But not all of them, just some of it, usually on those devices:
- Samsung s20 - Android 13
- Samsung a52 - Android 13
- Samsung Z fold - Android 13

So it fail as of Fail reason -100 and BLE get disconnected. And it always occurs before last or last few byte transferred. 

Here is some filtered Bluetooth logs if it can helps figuring out the issue. 

`
16:00:23.359 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 19088, appName: com.bellatrix.iweech
16:00:23.365 BtGatt.GattService      com.android.bluetooth       D  onScannerRegistered() - UUID=d6db85bb-5aa6-442f-8463-d10d87917ead, scannerId=10, status=0
16:00:23.383 BtGatt.GattService      com.android.bluetooth       W  unfreezeScan() - No AppScanStats found with uid: 10340
16:00:23.387 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:00:23.387 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:00:23.399 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
16:00:23.399 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
16:00:24.686 BtGatt.GattService      com.android.bluetooth       D  onScanResult to scannerId: 10- eventType=0x1b, addressType=0, address=B827EB_4, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-60, periodicAdvInt=0x0
16:00:24.690 BtGatt.GattService      com.android.bluetooth       D  onScanResult to scannerId: 10- eventType=0x1b, addressType=0, address=B827EB_4, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-60, periodicAdvInt=0x0
16:00:24.718 BtGatt.ContextMap       com.android.bluetooth       D  remove() - id: 10
16:00:24.718 BtGatt.ContextMap       com.android.bluetooth       E  remove() - removed: 10
16:00:24.746 BtGatt.GattService      com.android.bluetooth       D  registerClient(com.bellatrix.iweech) - UUID=5be2b9e1-5e17-4d78-be4f-efb12b83e38b
16:00:24.748 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 19088, appName: com.bellatrix.iweech
16:00:24.749 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:00:24.749 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:00:24.754 BtGatt.GattService      com.android.bluetooth       D  onClientRegistered() - UUID=5be2b9e1-5e17-4d78-be4f-efb12b83e38b, clientIf=7
16:00:24.763 BtGatt.GattService      com.android.bluetooth       D  clientConnect(com.bellatrix.iweech) - address = B827EB_4, isDirect=true transport =2 set own addr = false own addr type:0, clientIf: 7, opportunistic=false, phy: 1
16:00:25.074 BtGatt.GattService      com.android.bluetooth       D  onConnected() - clientIf=7, connId=7, status=0, address=B827EB_4
16:00:25.079 BtGatt.GattService      com.android.bluetooth       D  onConnected() - adding permit for address=B827EB_4
16:00:33.095 BtGatt.GattService      com.android.bluetooth       D  writeCharacteristic() - trying to acquire permit.
16:00:33.098 BtGatt.GattService      com.android.bluetooth       D  onWriteCharacteristic() - increasing permit for address=B827EB_4
16:00:33.099 BtGatt.GattService      com.android.bluetooth       D  onWriteCharacteristic() - address=B827EB_4, status=0
16:00:57.365 BtGatt.GattService      com.android.bluetooth       D  unregisterClient(com.android.bluetooth) - clientIf=7
16:00:57.366 BtGatt.ContextMap       com.android.bluetooth       D  remove() - id: 7
16:00:57.366 BtGatt.ContextMap       com.android.bluetooth       E  remove() - removed: 7
16:00:57.372 BtGatt.GattService      com.android.bluetooth       D  onDisconnected() - clientIf=7, connId=7, status=0, address=B827EB_4
16:00:57.372 BtGatt.ContextMap       com.android.bluetooth       E  Context not found for ID 7
16:00:57.372 BtGatt.GattService      com.android.bluetooth       D  onDisconnected() - removing permit for address=B827EB_4
16:01:02.999 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 10070, appName: com.bellatrix.iweech
16:01:03.000 BtGatt.GattService      com.android.bluetooth       D  onScannerRegistered() - UUID=b31a4d7f-ca66-4de0-bb76-19399ca45d4e, scannerId=10, status=0
16:01:03.014 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:01:03.014 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:01:03.016 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
[... cut ...]
16:01:22.838 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.google.uid.shared, scannerId: 2, reportDelayMillis=0
16:01:22.840 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
16:01:22.840 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
`

peropavlovic, 2023-02-02:
Hi @philips77 I still have this issue and trying to figure it out how to solve it. My users keep pushing me as they cannot use this major feature because of this issue.

So we found out it occurs on Samsung devices with Android 13. But not all of them, just some of it, usually on those devices:

Samsung s20 - Android 13
Samsung a52 - Android 13
Samsung Z fold - Android 13
So it fail as of Fail reason -100 and BLE get disconnected. And it always occurs before last or last few byte transferred.

Here is some filtered Bluetooth logs if it can helps figuring out the issue.

```

16:00:23.359 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 19088, appName: com.bellatrix.iweech
16:00:23.365 BtGatt.GattService      com.android.bluetooth       D  onScannerRegistered() - UUID=d6db85bb-5aa6-442f-8463-d10d87917ead, scannerId=10, status=0
16:00:23.383 BtGatt.GattService      com.android.bluetooth       W  unfreezeScan() - No AppScanStats found with uid: 10340
16:00:23.387 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:00:23.387 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:00:23.399 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
16:00:23.399 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_
16:00:24.686 BtGatt.GattService      com.android.bluetooth       D  onScanResult to scannerId: 10- eventType=0x1b, addressType=0, address=B827EB_4, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-60, periodicAdvInt=0x0
16:00:24.690 BtGatt.GattService      com.android.bluetooth       D  onScanResult to scannerId: 10- eventType=0x1b, addressType=0, address=B827EB_4, primaryPhy=1, secondaryPhy=0, advertisingSid=0xff, txPower=127, rssi=-60, periodicAdvInt=0x0
16:00:24.718 BtGatt.ContextMap       com.android.bluetooth       D  remove() - id: 10
16:00:24.718 BtGatt.ContextMap       com.android.bluetooth       E  remove() - removed: 10
16:00:24.746 BtGatt.GattService      com.android.bluetooth       D  registerClient(com.bellatrix.iweech) - UUID=5be2b9e1-5e17-4d78-be4f-efb12b83e38b
16:00:24.748 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 19088, appName: com.bellatrix.iweech
16:00:24.749 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:00:24.749 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_STOP, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:00:24.754 BtGatt.GattService      com.android.bluetooth       D  onClientRegistered() - UUID=5be2b9e1-5e17-4d78-be4f-efb12b83e38b, clientIf=7
16:00:24.763 BtGatt.GattService      com.android.bluetooth       D  clientConnect(com.bellatrix.iweech) - address = B827EB_4, isDirect=true transport =2 set own addr = false own addr type:0, clientIf: 7, opportunistic=false, phy: 1
16:00:25.074 BtGatt.GattService      com.android.bluetooth       D  onConnected() - clientIf=7, connId=7, status=0, address=B827EB_4
16:00:25.079 BtGatt.GattService      com.android.bluetooth       D  onConnected() - adding permit for address=B827EB_4
16:00:33.095 BtGatt.GattService      com.android.bluetooth       D  writeCharacteristic() - trying to acquire permit.
16:00:33.098 BtGatt.GattService      com.android.bluetooth       D  onWriteCharacteristic() - increasing permit for address=B827EB_4
16:00:33.099 BtGatt.GattService      com.android.bluetooth       D  onWriteCharacteristic() - address=B827EB_4, status=0
16:00:57.365 BtGatt.GattService      com.android.bluetooth       D  unregisterClient(com.android.bluetooth) - clientIf=7
16:00:57.366 BtGatt.ContextMap       com.android.bluetooth       D  remove() - id: 7
16:00:57.366 BtGatt.ContextMap       com.android.bluetooth       E  remove() - removed: 7
16:00:57.372 BtGatt.GattService      com.android.bluetooth       D  onDisconnected() - clientIf=7, connId=7, status=0, address=B827EB_4
16:00:57.372 BtGatt.ContextMap       com.android.bluetooth       E  Context not found for ID 7
16:00:57.372 BtGatt.GattService      com.android.bluetooth       D  onDisconnected() - removing permit for address=B827EB_4
16:01:02.999 BtGatt.ContextMap       com.android.bluetooth       D  add() - appUid: 10392, appPid: 10070, appName: com.bellatrix.iweech
16:01:03.000 BtGatt.GattService      com.android.bluetooth       D  onScannerRegistered() - UUID=b31a4d7f-ca66-4de0-bb76-19399ca45d4e, scannerId=10, status=0
16:01:03.014 BtGatt.GattService      com.android.bluetooth       W  checkToReportData before boot completed
16:01:03.014 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.bellatrix.iweech, scannerId: 10, reportDelayMillis=0
16:01:03.016 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
[... cut ...]
16:01:22.838 BtGatt.GattService      com.android.bluetooth       E  [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.google.uid.shared, scannerId: 2, reportDelayMillis=0
16:01:22.840 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : onFoundTimeout | onLostTimeout | onFoundCount | numOfTrackingEntries | scanFilterRssiThreshold | trackableScanFilterRssiThreshold | matchMode
16:01:22.840 BtGatt.ScanManager      com.android.bluetooth       D  configureFilterParamter() : __________500__|________10000__|___________1__|___________________0__|___________________-128__|____________________________-128__|________1_

```

Islomov49, 2023-02-13:
@peropavlovic have you solved this issue? I've faced with that as well

peropavlovic, 2023-03-13:
@adi-pilav-sn  thanks, I tried that.
@Islomov49 I did not find solution for the issue but I fixed it with different implementation on server side as a workaround. Its not perfect but its working for now.

johnnyzen, 2023-03-16:
@peropavlovic please can you share your fix?

deepak9705, 2023-05-22:
Can you please share the fix