nordicsemi/Android-BLE-Library #464: Example on how to read / use RSSI-Values

schattennarr, 2023-01-25:
Hi,

I'm looking for an example on how to use the readRssi()-Function exactly.
In my case I need the RSSI Value to sort for highest signal first in a space where many of our devices are available. From the documentation it's not quite clear how readRssi() tries to achieve this. I couldn't find anything in the trivia game for that too

philips77, 2023-01-25:
Mind, that this library is only for connected Bluetooth LE devices, not for scanning.

You may scan using native [`BluetoothLeScanner`](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner) (if targeting API 21+), or [Scanner Compat library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library), if you need similar API on older versions.

In any case you'll get a callback with RSSI for each scanner device.

philips77, 2023-01-25:
This method:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/cf95abb3742a1fdcfe3fd65e6fd29eb79508259f/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L2114-L2124
is only when you want to read RSSI of a connected device.

Then, you need to connect first, initialize it in a desired way, and you may enqueue that request to get RSSI updates. Keep in mind, that devices may adjust the TX power depending on the signal quality, so you may get unreliable values when in connection, as no matter the distance some devices try to keep the RSSI at similar level to save battery.

closed 2023-01-25 by philips77