nordicsemi/Android-BLE-Library #465: Data - getFloatValue() getting wrong conversion

Regjoz, 2023-01-30:
During unit testin I found that the float formated is output incorrect values.

I load the class Data with the bytearray (0xc1,0xbb,0x99,0x9a) y call getFloatValue with label FORMAT_FLOAT and the result is -0.000000 when should be -23,45.

I also tried changing the order of the bytes and same result

After checkng the implementation I think it will be more efficient to use java.nio.ByteBuffer to make all the transformation for int, longs and floats.

Loading the bytearray (0xc1,0xbb,0x99,0x9a) directly to ByteBuffer and calling getFloat I receive -23.45 (expected value)

philips77, 2023-01-31:
The issue may be Float encoding.
The `Data` class uses IEEE-11073 encoding, which you may find more about here: https://stackoverflow.com/questions/60841331/dart-convert-ieee-11073-32-bit-float-to-a-simple-double
The standard Float encoding uses IEEE-754 format (https://en.wikipedia.org/wiki/Single-precision_floating-point_format). `ByteBuffer` uses the latter. 

Format IEEE-11073 is popular in medical devices and in Bluetooth SIG adopted characteristics. It is also easier to decode, as the first byte is exponent and 3 remaining are the mantissa.

In your case, if you need IEEE-754, just use the `ByteBuffer`.

philips77, 2023-01-31:
In your example both 0xC1 and 0x9A are highly negative numbers in SInt8, so the exponent is -63 or -102. Anything to such power is pretty close to 0.

closed 2023-01-31 by philips77

philips77, 2023-01-31:
It kinda is, if you know where to look for :)
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/cf95abb3742a1fdcfe3fd65e6fd29eb79508259f/ble/src/main/java/no/nordicsemi/android/ble/data/Data.java#L190