nordicsemi/Android-BLE-Library #474: (BLE Server) sendNotification method fail callback not response

WanDa1993, 2023-03-02:
[Server]
```
            sendNotification(mNotifyCharacteristic, bytes, 0, bytes.length)
                    .before(new BeforeCallback() {
                        @Override
                        public void onRequestStarted(@NonNull BluetoothDevice device) {
                            log(Log.VERBOSE, "onRequestStarted");
                        }
                    })
                    .invalid(new InvalidRequestCallback() {
                        @Override
                        public void onInvalidRequest() {
                            log(Log.VERBOSE, "onInvalidRequest");
                        }
                    })
                    .fail(new FailCallback() {
                        @Override
                        public void onRequestFailed(@NonNull BluetoothDevice device, int status) {
                            log(Log.VERBOSE, "onRequestFailed");
                        }
                    })
                    .done(new SuccessCallback() {
                        @Override
                        public void onRequestCompleted(@NonNull BluetoothDevice device) {
                            log(Log.VERBOSE, "onRequestCompleted");
                        }
                    })
                    .then(new AfterCallback() {
                        @Override
                        public void onRequestFinished(@NonNull BluetoothDevice device) {
                            log(Log.VERBOSE, "onRequestFinished");
                        }
                    })
                    .enqueue();
```

reason：
BleManagerHandler#internalSendNotification  (999 line)

```
		if (value != null && value.length == 2 && value[0] != 0) {
			log(Log.VERBOSE, () -> "[Server] Sending " + (confirm ? "indication" : "notification") + " to " + serverCharacteristic.getUuid());
			boolean result;
			if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) {
				log(Log.DEBUG, () -> "[Server] gattServer.notifyCharacteristicChanged(" + serverCharacteristic.getUuid() +
						", confirm=" + confirm +
						", value=" + ParserUtils.parseDebug(data) + ")");
				return serverManager.getServer().notifyCharacteristicChanged(bluetoothDevice, serverCharacteristic, confirm, data) == BluetoothStatusCodes.SUCCESS;
			} else {
				log(Log.DEBUG, () -> "[Server] characteristic.setValue(" + ParserUtils.parseDebug(data) + ")");
				serverCharacteristic.setValue(data);
				log(Log.DEBUG, () -> "[Server] gattServer.notifyCharacteristicChanged(" + serverCharacteristic.getUuid() + ", confirm=" + confirm + ")");
				result = serverManager.getServer().notifyCharacteristicChanged(bluetoothDevice, serverCharacteristic, confirm);

				// The onNotificationSent callback is not called before Android Lollipop.
				if (result && Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP) {
					post(() -> {
						notifyNotificationSent(bluetoothDevice);
						nextRequest(true);
					});
				}
			}
			return result;
		}
		// Otherwise, assume the data was sent. The remote side has not registered for them.
		nextRequest(true);
		return true;
```

if value == null 

error：
1.notifyStarted

correct：
1.notifyStarted
2.notifyInvalidRequest or notifyFail

philips77, 2023-03-16:
Hi,
Sorry for the delay, I was busy with other projects. I've edited your question to make it more easy to understand. I hope I got it correctly.

You're right. Right now, if notifications were not enabled there is no callback, which indeed is weird. I will think what the outcome should be. It is wrong, as when you'd call `sendNotification(...).suspend()` the suspension would never finish.

Thank you for reporting the issue.

WanDa1993, 2023-03-16:
You're right.

PR #490 (2023-04-04): Sending a notification when not enabled returns a failure
This PR fixes #474.

Before the change, if notifications or indications were not enabled by client and the server tried to send one, the request was quietly ignored. This might have caused issues when using `suspend()`, as such suspension would never finish.
After the change such request will return in a failure with status `FailCallback.REASON_NOT_ENABLED`. `suspend()` or `await()` methods should throw `RequestFailedException` with that status.
closed 2023-04-04 by philips77