nordicsemi/Android-BLE-Library #478: Android 13 Bluetooth Disconnection - Pixel 4 issues

johnnyzen, 2023-03-14:
Hi all,

Phone: Pixel 4 Android 13 Feb 2023 update
Hardware: BlueMod+S42

I am getting a steady disconnection after subscribing to notifications:

enableNotifications(txCharacteristic).enqueue()

Seems like big issues around Android 13 and Bluetooth: https://issuetracker.google.com/issues/242755161

Some blood monitoring apps have managed to fix the issue (not sure if same issue), so am thinking its fixable.

Enabling notifications for 00000002-0000-1000-8000-008025000000 (2)
gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00) (3)
onClientConnectionState() - status=0 clientIf=6 device=00:80:25:E4:BD:FF
[Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED) (3)
Connection lost (5)

Looking at bluetooth snoop log I did the other day:

`03-09 15:48:01.146  1002 11431 11528 W bt_stack: [WARNING:btif_bqr.cc(390)] Appro LSTO , Handle: 0x0003, NULL, Central, PwLv: 9, RSSI: -56, SNR: 0, UnusedCh: 55, UnidealCh: 0, ReTx: 0, NoRX: 0, NAK: 0, FlowOff: 0, OverFlow: 0, UndFlow: 0
03-09 15:48:01.149  1002 11431 11528 I bt_stack: [INFO:btsnoop.cc(338)] clear_l2cap_allowlist: Clearing acceptlist from l2cap channel. conn_handle=3 cid=4:4
03-09 15:48:01.149  1002 11431 11528 I bt_stack: [INFO:gatt_attr.cc(338)] gatt_connect_cback: remove untrusted client status, bda=00:21:7e:4f:cf:ad
03-09 15:48:01.150  1002 11431 11528 I  bluetooth: system/bt/stack/gap/gap_ble.cc:298 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:cf:ad callback:Disconnected
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:3 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:4 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:5 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:6 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:7 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:8 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:9 transport:le reason:GATT_CONN_TIMEOUT
03-09 15:48:01.150  1002 11431 11528 W bt_btm_sec: system/bt/stack/btm/btm_sec.cc:3685 btm_sec_disconnected: Got uncommon disconnection reason:Connection Timeout handle:0x0003
03-09 15:48:01.150  1002 11431 11528 I btif_av : system/bt/btif/src/btif_av.cc:3352 btif_av_acl_disconnected: btif_av_acl_disconnected: Peer 00:21:7e:4f:cf:ad : ACL Disconnected
03-09 15:48:01.150  1002 11431 11528 I btif_av : system/bt/btif/src/btif_av.cc:1078 FindOrCreatePeer: BtifAvPeer *BtifAvSource::FindOrCreatePeer(const RawAddress &, tBTA_AV_HNDL): Create peer: peer_address=00:21:7e:4f:cf:ad bta_handle=0x41 peer_id=0
03-09 15:48:01.150  1002 11431 11528 I btif_av : system/bt/btif/src/btif_av.cc:3279 btif_av_get_peer_sep: No active sink or source peer found
03-09 15:48:01.150  1002 11431 11528 I bt_btif_a2dp_source: system/bt/btif/src/btif_a2dp_source.cc:691 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
03-09 15:48:01.150  1002 11431 11528 I btif_av : system/bt/btif/src/btif_av.cc:3279 btif_av_get_peer_sep: No active sink or source peer found
03-09 15:48:01.150  1002 11431 11528 I bt_btif_a2dp_source: system/bt/btif/src/btif_a2dp_source.cc:691 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
03-09 15:48:01.150  1002 11431 11528 I btif_av : system/bt/btif/src/btif_av.cc:1123 DeleteIdlePeers: DeleteIdlePeers: Deleting idle peer: 00:21:7e:4f:cf:ad bta_handle=0x41`

Any ideas?

menssen, 2023-04-02:
I just lost a weekend to this. Since you call out the February update specifically, and since the device you mention is bluetooth 4.2, your problem is probably the same as mine, which is this bug:

https://issuetracker.google.com/issues/262484178

which is fixed in this commit:

https://android.googlesource.com/platform/packages/modules/Bluetooth/+/9f5c4bc934c

which is included in the MARCH Pixel update (android-13.0.0_r35 / TQ2A.230305) but not in the FEBRUARY Pixel update (android-13.0.0_r30 / TQ1A.230205).

As far as I can tell, a workaround for before this fix was put in place is not possible. I tried decompiling versions 2.8.2 and 2.8.4 of that Abbot FreeStyle LibreLink blood app that supposedly fixed it and comparing, but I can't find anything that would affect this behavior. I'm wondering if their problem was something else.

I have given up, but maybe somebody else has an idea. Not sure how many other non-Google devices Android 13 made it to before that commit was incorporated, but according to the linked bug report, at least some Samsungs.

johnnyzen, 2023-04-03:
Hi @menssen Thanks for the link, although having to move onto a different project at the moment. One of the issues I have noticed, yet unable to investigate further due to other commitments was the inability to set the MTU for the size of the data I was sending, which resulted in a disconnect with GATT_CONN_TIMEOUT messages.

Have you investigated anything around the MTU?

johnnyzen, 2023-04-03:
I was able to get it working when sending a small packet a data under 23 bytes, which is the default MTU size.

philips77, 2023-09-29:
I'll close the issue as it has been stale now.

closed 2023-09-29 by philips77