nordicsemi/Android-BLE-Library #482: onDescriptorWrite error 128 , Error (0x80): GATT NO RESOURCES

doanpt, 2023-03-22:
My app is using your lib ver:2.5.1 and it works normally, but sometimes, my app(that run on android 13)  can not connect "lock device" with the error: 

> bt_stack: [WARNING:bta_gattc_act.cc(365)] bta_gattc_open_fail: Cannot establish Connection. conn_id=000000. Return GATT_ERROR(133)

After a few days research that issue, I think that is an android 13 BLE problem. https://issuetracker.google.com/issues/242755161?pli=1. 
On the other hand, I see the latest lib release is 2.6.0. so I decided to update my app with the latest version. After upgrade. I tested my app and I see the error below that was never found in ver 2.5.1:

> 03-22 21:10:23.451 29141 29141 D BleInterface: Notification received from xxxxxxxx-766c-42c4-8944-42bc98fc2d09, value: (0x).......
03-22 21:10:23.452 29141 29141 D BleInterface: setIndicationCallback disabling indication : xxxxxxxx-766c-42c4-8944-42bc98fc2d09
03-22 21:10:23.453 29141 29141 D BleInterface: gatt.setCharacteristicNotification(xxxxxxxx-766c-42c4-8944-42bc98fc2d09, false)
03-22 21:10:23.453 29141 29141 D BluetoothGatt: setCharacteristicNotification() - uuid: xxxxxxxx-766c-42c4-8944-42bc98fc2d09 enable: false
03-22 21:10:23.459 29141 29141 D BleInterface: Disabling notifications and indications for xxxxxxxx-766c-42c4-8944-42bc98fc2d09
03-22 21:10:23.459 29141 29141 D BleInterface: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x00-00)
03-22 21:10:23.465  2571  2985 I bt_stack: [INFO:gatt_utils.cc(1470)] gatt_send_write_msg len=2
03-22 21:10:23.465  2571  2985 I bluetooth: attp_cl_send_cmd: Sending ATT command to l2cap cid:0x0004 eatt_channels:0 transport:BT_TRANSPORT_LE
03-22 21:10:23.466  2571  2985 I bluetooth: attp_cl_send_cmd: Starting ATT response timer
03-22 21:10:23.544  2571  2985 I bt_stack: [INFO:gatt_cl.cc(1188)] gatt_client_handle_server_rsp opcode: 0x01
03-22 21:10:23.545  2571  2985 I bt_stack: [INFO:gatt_cl.cc(533)] gatt_process_error_rsp
03-22 21:10:23.545  2571  2985 I bt_stack: [INFO:gatt_utils.cc(1500)] gatt_end_operation status=128 op=3 subtype=2
03-22 21:10:23.547 29141 29141 E BleManager: onDescriptorWrite error 128
03-22 21:10:23.548 29141 29141 D BleInterface: Error (0x80): GATT NO RESOURCES
03-22 21:10:23.549 29141 29141 D BleInterface: Disabled BLE indication : 4d050051-766c-42c4-8944-42bc98fc2d09

But after facing the error, my app still works normally. So could you please let me know what happen with my error above? Thanks

philips77, 2023-03-22:
Regarding the first issue, when you do `retry(times, delay)` it should retry to connect if you're also getting this error in the library and it can handle it.

The other issue, is it present after restarting Bluetooth? How do you reproduce it?

doanpt, 2023-03-23:
for "onDescriptorWrite error 128 , Error (0x80): GATT NO RESOURCES"
I'm facing it sometimes, my app has many steps that need to be done before app can communicate with device as below:
1. generate Digital Signature
2. verify Digital Signature
3. verifyDevice Cert
4. send Key And Salt
5. add access codes

In each step, we call the methods:
SetIndicationCallback
EnableIndication
WriteCommand

Then I faced the error after step 4.