nordicsemi/Android-BLE-Library #486: Question: Auto Connect

chevamax, 2023-03-30:
Hi,
First of all, thanks for this awesome library! 

I'm having trouble understanding the auto connect feature / the ConnectionState.

Some context: I scan for a device, call connect( with `AutoConnect` set to true), then bond with the device. As I need to be connected to the device when it is reachable, I thought that having the system handle this for me would be nice and not as battery consuming as if I do the scan myself.

The following behaviour happens when using autoConnect to the bonded device or not:

I use the `BleManager.stateAsFlow()` to observe connection updates. When connecting to the device with `AutoConnect` set to true, on first disconnection I can see a **"GATT CONN TIMEOUT"** and a **"Connection lost"** warning log. Then the `ConnectionState` received is "Connecting" (there is no Disconnected state).
Once the device is in range, it reconnect as expected.
The second time (and afterward) that the device disconnect, I can see the same warning message as the first time, but the ConnectionState received is now "Disconnected(reason=LINK_LOSS)".
Again, once the device is in range, it reconnect as expected.

I'd like to understand why the 2nd and following disconnections are giving a "Disconnected" `ConnectionState`, as I would like after the disconnection, to know if the system will reconnect to the device on its own or if I should call again the connect method. I did not see how to check if BluetoothGatt is registered or not from code (but I can see that in the logcat).

Maybe I used the wrong approach with this ?

Thanks,

philips77, 2023-04-04:
I managed to replicate the issue:
```
13:43:17.149 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Connecting@c1b8cd5
13:43:17.493 BluetoothGatt           no.nordicsemi.android.nrfblinky      D  onClientConnectionState() - status=0 clientIf=7 device=D7:6E:14:18:2D:30
13:43:17.501 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
13:43:17.517 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Initializing@90d6055
13:43:18.583 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Ready@615bb5b
13:43:47.009 BluetoothGatt           no.nordicsemi.android.nrfblinky      D  onClientConnectionState() - status=0 clientIf=7 device=D7:6E:14:18:2D:30
13:43:47.011 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
13:43:47.046 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Connecting@c1b8cd5
13:43:54.706 BluetoothGatt           no.nordicsemi.android.nrfblinky      D  onClientConnectionState() - status=0 clientIf=7 device=D7:6E:14:18:2D:30
13:43:54.707 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
13:43:54.737 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Initializing@90d6055
13:43:55.306 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: no.nordicsemi.android.ble.ktx.state.ConnectionState$Ready@615bb5b
13:44:07.213 BluetoothGatt           no.nordicsemi.android.nrfblinky      D  onClientConnectionState() - status=0 clientIf=7 device=D7:6E:14:18:2D:30
13:44:07.215 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
13:44:07.245 BlinkyManagerImpl       no.nordicsemi.android.nrfblinky      I  State: Disconnected(reason=LINK_LOSS)
```

chevamax, 2023-04-04:
Yeah that's exactly the behaviour I have ! (sorry for not providing logs, I should have)

For now I check the disconnection reason to know if I need to reconnect (if the reason is LINK_LOSS I assume that the system will reconnect for me).

philips77, 2023-04-04:
Have a look at proposed solution. With that change I'm getting all states successfully.

PR #489 (2023-04-04): Switching `stateAsFlow` and `bongingStateAsFlow` from StateFlow to SharedFlow
This PR fixes #486.

The reason why `Disconnected` event was not caught using `stateAsFlow` on first connection when using `autoConnect = true` was that in this case the library was immediately starting reconnecting to the target peripheral using `gatt.connect()`. Before that, a new event `Connecting` was sent which was overwriting the state of the manager. As `StateFlow` doesn't have any buffering, a value that was not consumed is overwritten and was lost.
In case of using `autoConnect = false` or in following conne [... cut]
closed 2023-04-04 by philips77