nordicsemi/Android-BLE-Library #495: Incorrect bond state of already bonded BluetoothDevice

david-gettins, 2023-05-06:
I would like to be able to press a button in an app and bond with a BLE device. Next time I open the app and reconect the BLE device I would like to see if it is already bonded and hide the button. Unfortuntely the `ObservableBleManager` `bondingState` is incorrectly reported as `NOT_BONDED` when it actually is bonded. Android's `BluetoothDevice` shows it as being bonded.

*To Reproduce*

1. Connect to a BLE device
2. Call `ensureBond().enqueue()`
3. Disconnect BLE device
4. Close and re-open app
5. Connect BLE device
6. `BondState` is `NOT_BONDED` but `bluetoothDevice.bondState == BluetoothDevice.BOND_BONDED` the Android OS reports it as bonded also.

*Notes*

After bonding for the first time, `BondState` is correct. `BondState` is also correct if I remove bond and ensureBond again.

philips77, 2023-05-15:
I'll check that soon. Thank you for raising the issue

roopeshtek, 2023-05-26:
We are also facing similar issue on our Android 13 devices. Its not connecting with BLE peripherals. Do you have any insight on this? We dont see  google is working on it as the issue is open more than 6 months. 

https://issuetracker.google.com/issues/242755161

BurgerSzilard, 2023-11-05:
Any news on this ? Will this be fixed / is this fixed in one of the new releases ?

philips77, 2023-11-05:
Sorry, I missed that. I'll come back to you soon.