nordicsemi/Android-BLE-Library #502: is there a react-native library ?

cn-johndpope, 2023-06-20:
is there any plans to build this? will ble work with react-native? any guidance would be good.

cn-johndpope, 2023-06-20:
[image]

can see there are some libraries - but not related to BLE.
We want some guidance to use the java code from react.

cn-johndpope, 2023-06-27:
@philips77 ?

philips77, 2023-06-28:
Hi,
We do not support react-native. Some other folks have made a wrapper for our DFU libraries: https://github.com/Salt-PepperEngineering/react-native-nordic-dfu but that's all sorcery for me. You may find that link useful.