nordicsemi/Android-BLE-Library #503: iOS devices automatically disconnects after connecting

worstkiller, 2023-06-22:
Hey All, 

I'm facing an issue with the BLE connection. So what exactly happening is that some iOS-based clients are trying to connect with Android-based advertisers and we are using this library to do the job and it works in most of our use cases. 
``` implementation 'no.nordicsemi.android:ble-ktx:2.6.0'```
Except for few times when some iOS-based devices are trying to connect and it connects and just after that it disconnects automatically.

So here is the sequence of overridden functions that gets called.
- `onDeviceConnectedToServer(device: BluetoothDevice)` \\device connects
- `it does some job and writes back using ClientConnection which extends BleManager` \\does the job and writes back
- `onDeviceDisconnectedFromServer(device: BluetoothDevice) ` \\disconnects
- `onServicesInvalidated() ` \\invalidates the server connection

It is able to do the job on repeat but in some cases, it does not does the main job of writing back the data as before writing anything it gets disconnected as soon as it connects.
- `onDeviceConnectedToServer(device: BluetoothDevice)` \\device connects
- `onDeviceDisconnectedFromServer(device: BluetoothDevice) ` \\disconnects
- `onServicesInvalidated() `  \\invalidates the server connection

the issue is that it disconnects just after it connects and it is doing continuously so for some specific devices. But it is working fine for other ios devices and Android devices.

And since we are not getting any error msg or status code so we are not sure what's causing the issue. If you guys are aware of anything related to this issue, please help us or any suggestions that we could do in config or setup from the advertiser side. OR should we look into the client side, please do suggest.

Thanks for library

DmytroBatyuk, 2023-07-03:
Observing the same issue. Any ideas how to fix this?

philips77, 2023-09-29:
Hi, sorry for the late response. Did you manage do fix the issue?
Do you have any logs which could be useful? They are sent to `log(...)` method in `BleManager`.