nordicsemi/Android-BLE-Library #504: onDeviceDisconnected - inconsistent reason

tamasimecs, 2023-07-04:
I'm getting different reasons inside the `ConnectionObserver`s `onDeviceDisconnected` callback for different devices in case of disconnecting some BLE devices from the phone (disconnecting them from the power outlet). 
OnePlus 5: Android 9 -> REASON_TIMEOUT
Samsung Galaxy S21 FE: Android 13 -> REASON_SUCCESS
Google Pixel 3: Android 12 -> REASON_SUCCESS

The main issue with this is that if I disconnect from a device using the `disconnect()` from the `BleManager`, I also get a REASON_SUCCESS on the `onDeviceDisconnected` callback.

Maybe the different Android version is the issue, I'm not sure. Can you help me take a look at it, how to solve it? 

Thanks,
Tamas

krioru, 2023-09-22:
I have the same issue.

I have a battery powered peripheral/server device and I use phone to connect to it and read its characteristics. If I remove device's battery, I get the `SUCCESS` reason on disconnection event.

If I try to connect to device which is not powered or out of reach, i get the `UNKNOWN` reason on disconnection event.

philips77, 2023-09-27:
I'll look into that, thanks.

philips77, 2023-09-27:
The intended reason should be `TERMINATE_PEER_USER`, I believe, when you kill the connected device. In worst case `TIMEOUT`.

krioru, 2023-09-27:
Removing the battery and going out of reach both give `SUCCESS` reason on my Android 13 phone.
In both cases the reason should be `TIMEOUT`, because it's not a 'graceful disconnection'.

philips77, 2023-09-27:
There is this method called at some point:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/839cbb3ac11b6b03d0dbc57f5aa523e19805ef62/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2810-L2820
If the reported status is 0, then success is reported. I'll try to modify it so SUCCESS is only reported when the disconnection was triggered by the user.

philips77, 2023-09-27:
Btw, what messages do you get on WARNING level just before getting the event from the `log(...)` method?

krioru, 2023-09-27:
I don't see any warning logs. I retrieve the reason from `stateAsFlow()` flow.

Here's the logs that I receive when I remove power from the peripheral device:

```
BluetoothGatt           D  onClientConnectionState() - status=0 clientIf=12 device=CC86EC_3
BluetoothC...Connection D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
BluetoothC...Connection I  Disconnected
BluetoothC...Connection D  gatt.close()
BluetoothGatt           D  close()
BluetoothGatt           D  unregisterApp() - mClientIf=12
BluetoothAdapter        I  STATE_ON
BluetoothAdapter        I  STATE_ON
BluetoothAdapter        I  STATE_ON
BluetoothC...Connection D  onServicesInvalidated()
BluetoothC...er:connect D  Disconnect reason: SUCCESS
```

The last log line is called whenever `stateAsFlow()` emits a new value.

Like this:

```kotlin
stateJob = bleConnectionManager?.stateAsFlow()?.onEach { managerState ->

  if (managerState is ConnectionState.Disconnected) {
    Log.d("$TAG:connect", "Disconnect reason: ${managerState.reason}")
  }
  
}?.cancellable()?.launchIn(scope)
```

krioru, 2023-09-27:
Ok, I checked on two other devices running Android 10 and Android 8 and they _do_ have that warning log and they report a correct reason:

```
D  [Callback] Connection state changed with status: 8 and new state: 0 (DISCONNECTED)
W  Error: (0x8): GATT CONN TIMEOUT
I  Disconnected
D  gatt.close()
D  onServicesInvalidated()
D  Disconnect reason: TIMEOUT
```

PR #522 (2023-09-29): Disconnection reason and events fixed
This PR fixes #504.

Disconnection result `RESULT_SUCCESS` should now only be returned when the user has requested disconnection using `disconnect()` method. When the connected device goes out or range or is turned off the `REASON_TIMEOUT` should always be returned.
closed 2023-09-29 by philips77