nordicsemi/Android-BLE-Library #507: Need BLUETOOTH PRIVILEGED permission - Android 13

archie94, 2023-07-24:
On certain user devices (all Android 13) we are facing the following crash. 

```
Fatal Exception: java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10145 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
       at android.app.ContextImpl.enforce(ContextImpl.java:2240)
       at android.app.ContextImpl.enforceCallingOrSelfPermission(ContextImpl.java:2268)
       at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:948)
       at com.android.bluetooth.Utils.enforceBluetoothPrivilegedPermission(Utils.java:388)
       at com.android.bluetooth.gatt.GattService.permissionCheck(GattService.java:513)
       at com.android.bluetooth.gatt.GattService.readCharacteristic(GattService.java:3789)
       at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.readCharacteristic(GattService.java:987)
       at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.readCharacteristic(GattService.java:975)
       at android.bluetooth.IBluetoothGatt$Stub.onTransact(IBluetoothGatt.java:970)
       at android.os.Binder.execTransactInternal(Binder.java:1285)
       at android.os.Binder.execTransact(Binder.java:1244)
```

[Something related I could find on SF](https://stackoverflow.com/questions/76153874/android-13-need-bluetooth-privileged-permission/76226902#76226902) but we are not enabling notification on any restricted characteristics. 
Is there any way the library can shed more info on these crashes?

LuoPeiQin, 2023-07-24:
I also encountered the same problem.

philips77, 2023-07-24:
Hello, what characteristic you're trying to read? Are you sure you have the correct UUID and reference?

archie94, 2023-07-24:
Hello!

Our peripheral has few standard services like `0000180a-0000-1000-8000-00805f9b34fb` and `0000181e-0000-1000-8000-00805f9b34fb` but the rest are our custom services. 

Hex code are `86f65000-f706-58a0-95b2-1fb9261e4dc7` / `86f66000-f706-58a0-95b2-1fb9261e4dc7` / `86f61000-f706-58a0-95b2-1fb9261e4dc7` 

This crash occurs on android 13 devices only so not sure if this has to do with wrong implementation on our end.

archie94, 2023-07-24:
Unfortunately this we haven't found in our test devices. All on remote user devices. Anything else you think we can try?

corentin-c, 2023-07-25:
I think I remember having this exception when the bluetooth is disabled on some devices. Did you make sure bluetooth is enabled before doing those calls ?

archie94, 2023-07-25:
Yes. Saw the same on a stackoverflow thread. We make sure bluetooth is enabled before connecting to a peripheral.

plotczyktest, 2023-08-09:
We also encounter this issue.
I think that this is somehow Android 13 related as all those errors occur on this particular version.
The problem may be that Android i caching characteristics for particular BLE devices.
Like [NoviaWu123](https://github.com/NoviaWu123) mentioned if for some reason we first make a connection with device that uses forbidden characteristics and later connect with second device not using any forbidden characteristics it may happen that Android will share this connection ID and aggregate characteristics from both of those devices (not sure how it may happen)
[GattService.java](https://cs.android.com/android/platform/superproject/main/+/refs/heads/main:packages/modules/Bluetooth/android/app/src/com/android/bluetooth/gatt/GattService.java;drc=75e2aaa36534ebf4561f886c0372dba6fac4c852;l=2438)

The only solution I can think of is to try refreshing deviceChache (refreshDeviceCache()) and then reconnecting.
However we would need a trigger for it like catching this particular exception `java.lang.SecurityException`
This exception however can only be caught inside Android-BLE-Library

So my suggestion would be to  add a try catch inside [BleManagerHandler#internalReadCharacteristic()](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/004e6645dfb391c253c13a8b75d38a86b0aeacc2/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L1054C21-L1054C21) and possibly add new [FailCallback reason](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/main/ble/src/main/java/no/nordicsemi/android/ble/callback/FailCallback.java) that will be notified to Android App (without crashing the whole app) so that devs can handle this exception them self like for example calling refreshDeviceCache() and trying to reconnect

nadrolinux, 2023-08-11:
I also see this issue on Pixel 6 with Android 13, however it starts today, yesterday all was fine. After phone reboot all starts working again. I think it may be related to connected smartband (Xiaomi Band 6) to Pixel 6 , because when smartband is disconnected all works fine. I also see that when smartband is connected it auto connect to my gatt server dedicated to my app and I'm not sure how can I filter it.

elongpaoxiao, 2023-09-05:
The result of my testing here is that if I use a Google Android 13 phone to connect, the previous connected device contains a special uuid defined by Google, and I connect to the next device that does not contain a special uuid, this problem will occur.
<html>
<body>

private static final UUID HID_SERVICE_UUID =
--
  | UUID.fromString("00001812-0000-1000-8000-00805F9B34FB");
  |  
  | private static final UUID[] HID_UUIDS = {
  | UUID.fromString("00002A4A-0000-1000-8000-00805F9B34FB"),
  | UUID.fromString("00002A4B-0000-1000-8000-00805F9B34FB"),
  | UUID.fromString("00002A4C-0000-1000-8000-00805F9B34FB"),
  | UUID.fromString("00002A4D-0000-1000-8000-00805F9B34FB")
  | };
  |  
  | private static final UUID ANDROID_TV_REMOTE_SERVICE_UUID =
  | UUID.fromString("AB5E0001-5A21-4F05-BC7D-AF01F617B664");
  |  
  | private static final UUID FIDO_SERVICE_UUID =
  | UUID.fromString("0000FFFD-0000-1000-8000-00805F9B34FB"); // U2F
  |  
  | private static final UUID[] LE_AUDIO_SERVICE_UUIDS = {
  | UUID.fromString("00001844-0000-1000-8000-00805F9B34FB"), // VCS
  | UUID.fromString("00001845-0000-1000-8000-00805F9B34FB"), // VOCS
  | UUID.fromString("00001843-0000-1000-8000-00805F9B34FB"), // AICS
  | UUID.fromString("00001850-0000-1000-8000-00805F9B34FB"), // PACS
  | UUID.fromString("0000184E-0000-1000-8000-00805F9B34FB"), // ASCS
  | UUID.fromString("0000184F-0000-1000-8000-00805F9B34FB"), // BASS
  | };

</body>
</html>

archie94, 2023-09-05:
Hi @philips77 
Any suggestions how to handle this issue?

archie94, 2023-09-05:
Others are you able to circumvent the problem?

plotczyktest, 2023-09-05:
Looks like toggling Bluetooth helps to resolve this issue.
After reanabling Bluetooth looks like device cache is somehow cleared and forbidden characteristics are removed from services the real appliance is providing (tested on Pixel 6)

However we still get this crash and cannot catch it outside of the "Android BLE library"
Can we ask for catching those inside of the lib and calling FailCallback with new status REASON?
`int REASON_BLUETOOTH_PERMISSION_ PRIVILEGED = -200;`
or smth like that?

plotczyktest, 2023-09-12:
We do not start Bluetooth communication when Bluetooth is Off :).
I only mentioned that restarting Bluetooth on the phone solves this issue. But crash still occurs.

philips77, 2023-09-25:
I'll see what I can do.

closed 2023-09-25 by philips77

archie94, 2023-12-26:
This crash seems to occur on 2.7.2 release also!

archie94, 2024-02-23:
Hello @philips77 any idea on this? 

This is the crashlog

```
Fatal Exception: java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10412 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
       at android.app.ContextImpl.enforce(ContextImpl.java:2332)
       at android.app.ContextImpl.enforceCallingOrSelfPermission(ContextImpl.java:2360)
       at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:982)
       at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:982)
       at com.android.bluetooth.Utils.enforceBluetoothPrivilegedPermission(Utils.java:477)
       at com.android.bluetooth.gatt.GattService.permissionCheck(GattService.java:474)
       at com.android.bluetooth.gatt.GattService.registerForNotification(GattService.java:4050)
       at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1119)
       at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1107)
       at android.bluetooth.IBluetoothGatt$Stub.onTransact(IBluetoothGatt.java:1088)
       at android.os.Binder.execTransactInternal(Binder.java:1363)
       at android.os.Binder.execTransact(Binder.java:1304)
```

@LuoPeiQin @plotczyktest @elongpaoxiao @nadrolinux @NoviaWu123 anyone still facing this issue?

philips77, 2024-02-23:
What characteristic are you trying enable notifications on? Is it any of those from here: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/507#issuecomment-1706118207 ?
Are you using version 2.7.3? I can't see any library method in your stack trace.

archie94, 2024-02-24:
Characteristics are same as i mentioned [here](https://github.com/NordicSemiconductor/Android-BLE-Library/issues/507#issuecomment-1647765435). 
We are on 2.7.2 
I see 2.7.3 has https://github.com/NordicSemiconductor/Android-BLE-Library/pull/545 which may fix the issue. Will check. Thanks!

archie94, 2024-03-21:
Although frequency has reduced a bit but the problem still remains. 

have created a new issue here: https://issuetracker.google.com/issues/330663537
earlier one was closed for inactivity here: https://issuetracker.google.com/issues/304893289

archie94, 2024-03-21:
I have sent a PR for handling this issue. 
@philips77 please review once you get time.

j0bro, 2024-05-14:
> Although frequency has reduced a bit but the problem still remains.
> 
> have created a new issue here: [issuetracker.google.com/issues/330663537](https://issuetracker.google.com/issues/330663537) earlier one was closed for inactivity here: [issuetracker.google.com/issues/304893289](https://issuetracker.google.com/issues/304893289)

Great to see, and thanks for re-reporting. Google admits it's on their side and a fix is planned for rollout to the field they say in the new issue.

kle-wang, 2024-08-19:
Although google admits and repair in android 15 beta3,most of user are not keep latest android version.
How can we do?I just notice catch error,but nothing to do. May be refresh gatt cache?

PR #518 (2023-09-25): Catching read and write errors
This PR fixes #507.

Reading or writing characteristics or descriptors with some restricted UUIDs may throw `SecurityExeception`. 
This PR adds a try-catch block. The error returned to the user is `FailCallback.REASON_NULL_ATTRIBUTE`.