nordicsemi/Android-BLE-Library #512: Android is moving bluetooth scan client to opportunistic more than it should

rickymohk, 2023-09-18:
I am developing an Android app to continuously scan for BLE devices.

First I encounter this problem https://stackoverflow.com/questions/43833904/android-bluetooth-le-scanner-stops-after-a-time

So I implement a timer to restart the scan at regular interval. But I still find it happening sometimes. When I dig into the Android source code, I found a subtle difference in the latest version of ScanManager.java.

In [Android 7](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/android-7.0.0_r1/src/com/android/bluetooth/gatt/ScanManager.java) :
```java
...
                    if (!mScanNative.isFirstMatchScanClient(client)) {
                        Message msg = mHandler.obtainMessage(MSG_SCAN_TIMEOUT);
                        msg.obj = client;
                        // Only one timeout message should exist at any time
                        mHandler.removeMessages(SCAN_TIMEOUT_MS);
                        mHandler.sendMessageDelayed(msg, SCAN_TIMEOUT_MS);
                    }
...
```

In [latest version](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/master/src/com/android/bluetooth/gatt/ScanManager.java#280):
```java
...
                    if (!mScanNative.isExemptFromScanDowngrade(client)) {
                        Message msg = obtainMessage(MSG_SCAN_TIMEOUT);
                        msg.obj = client;
                        // Only one timeout message should exist at any time
                        sendMessageDelayed(msg, AppScanStats.getScanTimeoutMillis());
                    }
...
```

The `removeMessage` call is removed inside `handleStartScan`. The `MSG_SCAN_TIMEOUT` message can now only be removed in `handleStopScan`. 

I suspect when I kill my app, I cannot call `stopScan` since my app is killed. I also lost the reference to my `scanCallback` so there is no way for me to call `stopScan` in new app session. Hence there is no way to cancel the `MSG_SCAN_TIMEOUT` previously set up. When I start my app again and obtain the BLE scanner with the same `scannerId`, the persisting `MSG_SCAN_TIMEOUT` will move my newly started scan to opportunistic. Am I correct about this? If this is the case, is this a bug in Android OS and is there any workaround I can do?

philips77, 2023-09-25:
Scanning is not in scope of this library. I recommend asking this question on StackOverflow.

As a quick answer I suggest using scan with a `PendingIntent`. This is the desired way to handle continuous scan on Android 8+. [Here](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)) is the API.

rickymohk, 2023-09-25:
Oops, sorry, I mixed it up. Should have asked this in [Android-Scanner-Compat-Library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library)