nordicsemi/Android-BLE-Library #526: Devices are not reconnecting after upgrade from 2.6.1 to 2.7.0

corentin-c, 2023-10-17:
Hi,

After we upgraded from 2.6.1 to 2.7.0, when using connect with autoConnect true,  devices are not reconnecting  after the first disconnection.

We use client and server with `clientManager.useServer(serverManager)`
After debugging what I understand is happening in our code is :

- In 2.6.1 : 
On disconnection, **server** manager `onConnectionStateChanged `is triggered
It **does not call** `notifyDeviceDisconnected `in this version
**client** manager `onConnectionStateChanged `is triggered. 
This set the variable `wasConnected `to the value of `connected `which is **true**
**calls** `internalConnect `because `wasConnected `is true and that will make the device autoconnect in the future

- In 2.7.0
On disconnection, **server** manager `onConnectionStateChanged `is triggered
This **now calls** `notifyDeviceDisconnected `that sets `connected `to false
**client** manager `onConnectionStateChanged `is triggered. 
This set the variable `wasConnected `to the value of `connected `which is **false**
**Does not call** `internalConnect `because `wasConnected `is false and that **will not** make the device autoconnect in the future

Is this a bug, or are we using the library wrong in some way ?

Thanks.

philips77, 2023-10-18:
Let me check. Thank you for the report.

philips77, 2023-10-18:
...and I'm not sure how to fix it. Need to think.

closed 2023-10-18 by philips77

corentin-c, 2023-10-19:
confirmed fixed on 2.7.1

PR #528 (2023-10-18): Reconnection fixed
This PR fixes #526.

I think this is minimal required change to fix the issue. The call to `notifyDeviceDisconnected(...)` was added to `BleServerManager` in #521. It was supposed to fix no disconnection callback in server-only mode when `attachClientConnection` is used.

However, as @corentin-c noticed, that broke bidirectional connection, as that method is now called twice, from server and client, and client doesn't reconnect.

The proposed check ensures that it is called only in server-only mode when `bluetoothGatt` object is null.