nordicsemi/Android-BLE-Library #547: Want to get the characteristic on which the data was received for the characteristics I have subscribed to get notified for

xaviergspintly, 2024-02-29:
I am subscribing to some characteristics using the `enableNotifications()` and I receive the data using a `DataReceivedCallback` .
That works fine.
However, I want to now also know on which particular characteristic object or characteristic UUID I'm getting the notification (meaning characteristic on which I'm getting the data in `DataReceivedCallback`).
Is there something similar to `DataReceivedCallback` which I can use to know this?

philips77, 2024-03-04:
You may register one callback per characteristic, so you'll know it by just having different implementations.

closed 2024-03-04 by philips77

xaviergspintly, 2024-03-04:
Yeah. I kind of did the same afterwards but forgot to close this issue. Thanks.