nordicsemi/Android-BLE-Library #549: Disconnect not working on various Samsung devices

niklasdahlheimer, 2024-03-18:
Thats probably more an issue for the Samsung support, but I wanted to get a few opinions here:

We had recent issue reports from various Samsung users:
Samsung S24 (SM-S921N) with Android 14 (SDK34)
Samsung S23 (SM-S911U1) with Android 14 (SDK34)

Their devices stay connected even if the disconnect via our app. 
The logs say, the `disconnect().done()`-callback as well as `onServicesInvalidated()` is called.

So I guess the devices are still connected on a lower BLE level.

I could NOT reproduce this issue with a S23 with Android 14 (Samsung firmware version S911BXXS3BXBD), so I guess it only 
affecting a subset of devices or a certain Samsung update.

Anyone else experience this issues?

philips77, 2024-03-18:
This may not be an issue at all. It depends.

The `BleManager`, just like the native `BluetoothGatt` which the manager is wrapping, are just virtual clients of the physical connection. The physical connection may be used by more than one client, both in a single app and in multiple apps, at the same time. One of such apps may be Android OS itself.

When you call `.disconnect().enqueue()`, this will only disconnect the one single client. It will receive DISCONNECTED event, as from its perspective, the connection is closed. If there are any other clients to the same device still connected, the physical link won't be affected. For example, Android OS may support some profiles natively, like HID. You may also have nRF Connect app (or similar) open and running in background, which can detect a new connection and create its own client to receive notifications and help with debugging.

It is only when the last client disconnects from the device the physical link is terminated as a result (approx after one second).

> [!TIP]
> To force closing a physical link to a Bluetooth LE device you may design a command which you may send to the device which will terminate the connection from its side. That way no matter how many clients were open on the phone, all of them will receive DISCONNECTED state.

niklasdahlheimer, 2024-03-18:
Thank you for this detailed information! I already ask the customers about nRF Connect app because I thought of this issue too, but it's not installed on their phones. But, yeah, maybe another app is getting in between and "holds" the connection. We will investigate on this. Thank you!