nordicsemi/Android-BLE-Library #552: No response from writeCharacteristic

Lightwood13, 2024-04-04:
Hi,
I think I found a bug in `writeCharacteristic` method. If BLE device disconnects right before the call to writeCharacteristic, then it never returns success or failure.

Example to reproduce:
```
disconnect()
    .done { Log.d("test", "Disconnect request successful") }
    .fail { _, _ -> Log.d("test", "Disconnect request failed") }
    .enqueue()

writeCharacteristic(
    communicationCharacteristic,
    byteArrayOf(),
    WRITE_TYPE_DEFAULT
)
    .done { Log.d("test", "Write request successful") }
    .fail { _, _ -> Log.d("test", "Write request failed") }
    .invalid { Log.d("test", "Write request invalid") }
    .enqueue()
```

This example prints in the logs `Disconnect request successful`, but write request neither succeeds nor fails.
I've also noticed the same behavior when BLE device disconnects because of poor signal and the code is trying to write characteristic at the same time.

When`writeCharacteristic` is used from kotlin with `.suspend()` like this:
```
writeCharacteristic(
    communicationCharacteristic,
    data,
    WRITE_TYPE_DEFAULT
).suspend()
```
it just hangs, never returning or throwing anything. Also, because `WriteRequest` is not cancellable when suspended, this coroutine can't be cancelled, so wrapping it in `withTimeout` doesn't help.

philips77, 2024-04-04:
I'll have a look, thank you for creating the issue.

dees91, 2024-06-21:
@philips77 Have you had a chance to check this topic? I have similar problem, if `writeCharacteristic` is called after `disconnect`, none of the `done`,`fail` or `invalid` callbacks are called, what's more, the `syncLock` underneath is never released, blocking the thread indefinitely.

philips77, 2024-06-21:
I didn't have time, sorry.

32penkin, 2024-08-02:
Hi guys, 
@Lightwood13 , @dees91 , have you found any solution for this issue?

Lightwood13, 2024-08-03:
@32penkin I'm using a workaround like this
```
suspend fun writeCharacteristic(
    characteristic: BluetoothGattCharacteristic,
    data: ByteArray,
    timeout: Duration
): Unit = withTimeoutOrNull(timeout) {
    suspendCancellableCoroutine { continuation ->
        writeCharacteristic(
            characteristic,
            data,
            WRITE_TYPE_DEFAULT
        )
            .done {
                if (!continuation.isCancelled) {
                    continuation.resume(Unit)
                }
            }
            .fail { _, status ->
                if (!continuation.isCancelled) {
                    continuation.resumeWithException(
                        BLERequestException("Write request failed with status $status")
                    )
                }
            }
            .enqueue()
    }
} ?: throw BLERequestException("Write request timed out")

class BLERequestException(message: String?) : Exception(message)
```
Here, instead of using library's `.suspend()` method of `WriteRequest`, that isn't cancellable, I'm manually suspending execution with `suspendCancellableCoroutine` and then call `.enqueue()` inside. If the write request doesn't succeed or fail within given timeout, the `withTimeoutOrNull` function will cancel the block inside it and return null, so we can handle this case. This method doesn't address the issue with lost resources and the thread that gets blocked indefinitely if request doesn't finish as @dees91 mentioned, but at least it allows to safely use `writeCharacteristic` from your code without it hanging.

philips77, 2024-08-29:
OK, I found the issue. When user requests disconnection, at some point the `close()` method is called, which clears task list without notifying them in any way.
I'll fix it asap.

philips77, 2024-08-29:
Hi,
I created a PR that seems to be fixing the issue with tasks not being notified.
I'll release it as a beta version and would be more than happy if you could try it.

closed 2024-08-29 by philips77

LightWeek, 2024-10-03:
Hi. I seems to be having the same or similar issue.
I am using version `2.9.0-beta02`
In my case i am calling 
```
try{
... some code BEFORE read
readCharacteristic(protectedCharacteristic).suspend()
... some code AFTER read
} catch(th: Throwable){
// handle error
}
```
After that the system pairing dialog appears (which is expected) and if you cancel the pairing request, the `readCharacteristic`  call just stays suspended forever, it doesn't throw, it does't continue. `... some code AFTER read` never called as well as catch block.

In logs i can see 
```
BleManager: Phone has lost bonding information
BluetoothGatt: close()
```
but readCharacteristic above just stays suspended.

If i enter the pin code instead of canceling - everything works, readCharacteristic continues and `... some code AFTER read` is called

philips77, 2024-10-04:
Thank you for diving into it!

Did you check the bond state receiver?
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/794e94fd9205dd8b8a2780bd6771abfb5b087cdb/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L334-L343
It should be called when bonding fails and should fail the awaiting request.

Some things I don't understand:
1. You're saying you get 
   > BleManager: Phone has lost bonding information

   printed in the logs, which is only logged when a request would fail with status 8 or 137 and bond state would not be `BOND_NONE`. 
2. Where is `gatt.close()` called. Is it called by the library or internally in Android (when bonding fails)?
3. Have a look at the same situation in nRF Connect for Android:
   [image]
   The app tries to read a characteristic, which triggers bonding. There's an 8 second gap where I click Cancel on the bonding popup and I get AUTH FAIL, like you do, but it's followed by the bond state broadcast (linked above), which should notify the request about a failure and, in effect, cancel suspended method.

LightWeek, 2024-10-04:
Ok, so actually 

> BleManager: Phone has lost bonding information

This message doesn't appear every time. But even when it appears `readCharacteristic(protectedCharacteristic).suspend()` still doesn't not throw any exception. 
I am debugging it now and the error is always 137 but `gatt.getDevice().getBondState()` in almost always =  BluetoothDevice.BOND_NONE

My code to connect looks something like this:
``` kotlin   
    suspend fun connect(scanResult: ScanResult): Result<MyDeviceData> {

        runCatching {
            connect(scanResult.device)
                .retry(3)
                .useAutoConnect(false)
                .timeout(DEVICE_CONNECTION_TIMEOUT_MILLIS)
                .suspend()
        }.onFailure {
            Log.e(TAG, "connect Connection failed", it)
        }

        Log.d(TAG, "connect Success. Connected Device = $bluetoothDevice")
        delay(200)
        val myDeviceData = runCatching {
            readCharacteristic(myDeviceCharacteristic).suspend()
        }.onFailure {
            Log.e(TAG, "connect Read myDeviceData failed", it)
        }

        Log.d(TAG, "connect myDeviceData = $myDeviceData")

        return myDeviceData
    }   
```
And `readCharacteristic(myDeviceCharacteristic).suspend()` here suspends forever if pin code dialog is canceled by user.
`connect` is called in `viewModelScope.launch{}`

Seems like some synchronization issue of bond state receiver. If i go slowly step by step in debuger, then it seems to be working sometimes, but if i remove breakpoints or run without debugger - it doesn't. 

Also, i have checked `mBondingBroadcastReceiver`
1.After initial connect/read characteristic request the pin dialog appears and state changes to **BOND_BONDING**.
2.When i click on `cancel` state changes to **BOND_NONE** 
3.almost immediately it changes to **BOND_BONDING** again (pin dialog shows up second time) 
4.I click `cancel` again and state changes to **BOND_NONE**

Step by step inside the receiver:
after first cancel request = null, so 
```if (request != null && (request.type == Request.Type.CREATE_BOND || request.type == Request.Type.ENSURE_BOND)```
is skipped.

```servicesDiscovered = false and serviceDiscoveryRequested = false```
so, as i understand start service discovery which triggers bonding again. (that's why pin dialogs shows up twice)

When i cancel the dialog second time 
the request.type = Request.Type.READ, so 
```if (request != null && (request.type == Request.Type.CREATE_BOND || request.type == Request.Type.ENSURE_BOND)```
is skipped again. 
request.notifyFail is not called

servicesDiscovered = true, serviceDiscoveryRequested = false (so it skipped as well this time)

LightWeek, 2024-10-04:
More info...

looks like I am getting 2 pairing dialogs because one is triggered by `connect(scanResult.device)...suspend()`, and the second one one by `readCharacteristic(myDeviceCharacteristic).suspend()`

And that's because 
```kotlin
connect(scanResult.device)
                .retry(3)
                .useAutoConnect(false)
                .timeout(DEVICE_CONNECTION_TIMEOUT_MILLIS)
                .suspend()
```
doesn't throw when pairing is canceled, it successfully continues (technically the device is connected, so no issue here), but why does it show popup during connection?

Then the sequential call to read characteristic triggers bonding again, the request type is `Request.Type.READ`, and then the request is not notified about the fail (details in previous message) after canceling the popup.
 
 Also attaching logs
 ```
17:47:50.388 BleConnectio...agerAndroid  D  scanForDevices scanResults:
                                            ScanResult{device=XX:XX:XX:XX:3A:EF, scanRecord=ScanRecord [advertiseFlags=6, serviceUuids=[00000001-07ce-44b0-9a00-efba512356e5, 8d53dc1d-1db7-4cd3-868b-8a527460aa84], manufacturerSpecificData=null, serviceData=null, txPowerLevel=-2147483648, deviceName=H206_953AEF], rssi=-75, timestampNanos=622747508227649, eventType=17, primaryPhy=1, secondaryPhy=0, advertisingSid=255, txPower=127, periodicAdvertisingInterval=0}
                                            ScanResult{device=XX:XX:XX:XX:B0:D7, scanRecord=ScanRecord [advertiseFlags=6, serviceUuids=[00000001-07ce-44b0-9a00-efba512356e5, 8d53dc1d-1db7-4cd3-868b-8a527460aa84], manufacturerSpecificData=null, serviceData=null, txPowerLevel=-2147483648, deviceName=H206_EEB0D7], rssi=-57, timestampNanos=622747540013171, eventType=17, primaryPhy=1, secondaryPhy=0, advertisingSid=255, txPower=127, periodicAdvertisingInterval=0}
17:47:50.831 BleConnectio...agerAndroid  D  awaitClose  scanner.stopScan(scanCallback)
17:47:50.848 BleConnectio...agerAndroid  D  connect device = BleDevice(address=C9:A4:71:EE:B0:D7, deviceName=H206_EEB0D7, pairedTime=2024-10-04T15:47:50.397166Z, lastConnected=2024-10-04T15:47:50.397176Z)
17:47:51.076 BleManager.log              V  Connecting...
17:47:51.077 BleManager.log              D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
17:47:51.078 BluetoothGatt               D  connect() - device: XX:XX:XX:XX:B0:D7, auto: false
17:47:51.078 BluetoothGatt               D  registerApp()
17:47:51.078 BluetoothGatt               D  registerApp() - UUID=c08e3fb9-41ea-43b6-9104-bcbad6276648
17:47:51.081 BluetoothGatt               D  onClientRegistered() - status=0 clientIf=13
17:47:51.203 BluetoothGatt               D  onClientConnectionState() - status=0 clientIf=13 device=C9:A4:71:EE:B0:D7
17:47:51.204 BleManager.log              D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
17:47:51.205 BleManager.log              I  Connected to C9:A4:71:EE:B0:D7
17:47:51.210 BleManager.log              D  wait(300)
17:47:51.245 BleManager.log              D  [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3); key: -1
17:47:51.252 BleManager.log              D  [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
17:47:51.710 BluetoothGatt               D  onPhyUpdate() - status=0 address=C9:A4:71:EE:B0:D7 txPhy=2 rxPhy=2
17:47:51.711 BleManager.log              I  PHY updated (TX: LE 2M, RX: LE 2M)
17:47:52.812 BleManager.log              D  [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
17:47:52.812 BleManager.log              W  Bonding failed
17:47:52.812 BleManager.log              V  Discovering services...
17:47:52.812 BleManager.log              D  gatt.discoverServices()
17:47:52.813 BluetoothGatt               D  discoverServices() - device: XX:XX:XX:XX:B0:D7
17:47:52.819 BluetoothGatt               D  onSearchComplete() = Device=C9:A4:71:EE:B0:D7 Status=0
17:47:52.822 BleManager.log              I  Services discovered
17:47:52.822 BleManager.log              V  Primary service found
17:47:52.824 BleManager.log              V  Requesting new MTU...
17:47:52.824 BleManager.log              D  gatt.requestMtu(498)
17:47:52.824 BluetoothGatt               D  configureMTU() - device: XX:XX:XX:XX:B0:D7 mtu: 498
17:47:52.880 BluetoothGatt               D  onConfigureMTU() - Device=C9:A4:71:EE:B0:D7 mtu=498 status=0
17:47:52.881 BleManager.log              I  MTU changed to: 498
17:47:52.882 BleConnectio...agerAndroid  D  connect Success. Connected Device = XX:XX:XX:XX:B0:D7
17:47:53.090 BleConnectio...agerAndroid  D  getBleDeviceInfo
17:47:53.097 BleManager.log              V  Reading characteristic 00000002-07ce-44b0-9a00-efba512356e5
17:47:53.098 BleManager.log              D  gatt.readCharacteristic(00000002-07ce-44b0-9a00-efba512356e5)
17:47:53.386 BleManager.log              D  [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
17:47:53.388 BleManager.log              D  [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_PIN (0); key: -1
17:47:55.146 BleManager.log              W  Authentication required (137)
17:47:55.149 BleManager                  W  Phone has lost bonding information
[... cut ...]
18:37:13.917 BluetoothGatt               D  unregisterApp() - mClientIf=13
```

Here we can see that after 
`18:37:10.865 BleManager.log              W  Bonding failed`

```kotlin
val myDeviceData = runCatching {
            readCharacteristic(myDeviceCharacteristic).suspend()
        }
```
this code is not suspended forever anymore and we can see the error in logs:
`18:37:10.869 BleConnectio...agerAndroid  D  connect deviceInfoResult = Failure(no.nordicsemi.android.ble.exception.RequestFailedException: Request failed with status -4)`

philips77, 2024-10-07:
Thank you for the detailed explanation! I'll look into it.

In the meantime, I wold recommend checking the following:
1. `getServiceDiscoveryDelay`
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/794e94fd9205dd8b8a2780bd6771abfb5b087cdb/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L649-L651 This method delays the service discovery after successful connection. You may try with returning 0 or increasing 300 ms to something longer. If your device always bonds just after connection, and the `bonded` param here is `false`, you may assume that the bonding process will start soon. The library initiates service discovery after this delay unless the device is in BONDING state (then it awaits bonding complete). This is to make sure the `connect` request does not finish until the device gets bonded.
2. You may also check bond state before calling `readCharacteristic` or try calling `createBond` request. It should succeed immediately when the device is already bonded.

philips77, 2024-10-07:
> To me seems legit to notify request about fail if bonding state changed from BONDING to BOND_NONE during request.type == Request.Type.READ (maybe WRITE as well?)

This should happen, yes. If you try to read a characteristic, which is protected, and the device initiates bonding, which fails, the read failed, so request should return an error (`suspend should throw). I'll look into why it's not the case.

LightWeek, 2024-10-07:
> In the meantime, I wold recommend checking the following:
>
> 1. `getServiceDiscoveryDelay` [...]

In my case i need to support 2 types of devices. 1. secured devices that can be bonded and 2. unsecured devices that only need to connect. In the 2nd case i don't need to request pairing or anything. And `createBondInsecure()` shows pairing dialog for the second case too, but without requesting pin code, just button to confirm. Which is, actually, OK.

I did not try `createBondInsecure()`/`createBond()` because it seemed insecure :)  
But now i see that if the characteristic is secured (we use lvl 4) it should not be an issue, correct?

So I'll stick with `createBondInsecure()` before calling characteristic for now. At least when it fails i can show correct error in the app, like "pin code required".

But still, would be ideal if you can check the issue with `readCharacteristic` request not being notified on bonding failure. 
The connection code i provided above is almost exact code i use for initial connection.
Test device : Pixel 7, Android 14
The BLE device is running on Zephyr OS.
We use security Level 4 for BLE 

The issue reproduces like 99% of connection attempts. Doesn't always reproduce in debug mode if breakpoints are set in `characteristicRead` and in the `broadcastreceiver` in the library code.

Also, when bonding is triggered by 
```kotlin
connect(connectToDevice)
                .retry(3)
                .useAutoConnect(false)
                .timeout(DEVICE_CONNECTION_TIMEOUT_MILLIS)
                .suspend()
``` 
shouldn't it also fail and throw an exception if bonding was canceled by user? Now it just continues successfully. It is quite ambiguous, it seems logical that it does not fail, and at the same time it seems that it is not.
 
Thanks!

LightWeek, 2024-10-07:
Few more observations. 
For new connection when i call 
```kotlin
private const val DEVICE_CONNECTION_TIMEOUT_MILLIS = 5_000L
connect(connectToDevice)
                .retry(3)
                .useAutoConnect(false)
                .timeout(DEVICE_CONNECTION_TIMEOUT_MILLIS)
                .suspend()
```
Android shows pairing popup with just 2 buttons "Pair" and "Cancel", also shows pairing notification in the notification tray, if I click "cancel" the `connect....suspend()` just resumes successfully.  Let's call this Case 1.
If I don't click anything then `connect....suspend()` fails with timeout. Or if I click "Pair" then another popup appears immediately, this time with Pin code field, but "connect....suspend()" fails with timeout before i finish entering pin code. 

But then, at some point when you accept first popup and cancel second one it starts only showing the second popup with pin code, without notification, etc... Even after re-installing the app/ rebooting the phone it keeps only showing the pin code popup. Let's call it Case 2.
At this point the device doesn't appear in bonded devices (yet) in phone settings. And in this case everything works ok. It doesn't fail with timeout while you entering pin code and if you cancel the popup the `connect....suspend()` immediately throws. Just like anyone would expect it to work. 
If you enter correct pin code and connect to this device, it appears in paired devices. Then if you unpair from system settings, then, on the next connection attempt, it will show the Case 1 again.

Case 1. Two Step pairing:
First popup, failed with timeout
[image]
[image]
If you click "pair" at this point the pin code popup will appear and you can still bond, but the app won't be notified about it as   `connect..suspend()` already thrown by timeout. if you click quickly "pair" or "pair and connect" in notification after it appears, `connect..suspend()` still will throw with timeout while you entering pin code in the pin code popup that appears.

Case 2. Only this dialog appears
[image]

But most often, especially for new connections Case 1 will be shown, Which can't be really handled if you set connection timeout. I would expect it to ignore connection request timeout while bonding and only return bonding result once canceled or succeed. And only throw timeout if it happens before bonding started. Case 1 can't be handled, because after timeout the device is disconnected. You only can try to connect again, which will fail again if user will type pin code longer than timeout (which is often around 5 seconds). For now the solution is to set big timeout or don't use it at all.

So, summing up, there are couple of issues.
1. Bonding during connection with timeout described above.
2. If "Pairing" popup is canceled (first popup from Case 1), `connect(device)....suspend()` resumes with no issues.
3. If case 2 happened, we assume connection successful. If in this case we read protected characteristic, which triggers bonding again.. and if bonding is canceled `readCharacteristic..suspend()` hangs.(Basically my original report)

philips77, 2024-10-08:
You may try disabling automatic bonding request from the device side, if possible, and request it from the phone, or just by reading this characteristic (setting MITM permission).

LightWeek, 2024-10-08:
Well, thank you for your answers. 
Ended up using only "connect" with timeout, and then if connection succeeds read protected characteristic + modified library code as on the screenshot.

[image]

Compared behaviour to NRF app, and seems to be working pretty much the same way. 
NRF app also shows 2 popups, one with pairing button only and second one for entering pin code. 

Tho don't really like it, because when it shows 2 popups that's tolerable, but when it comes to notifications only, well, i already see ton's of users having troubles with pairing the device. But couldn't find or come up with any solution that would work better.

Thanks for your help again. 
And will be waiting for the "forever suspended" readCharacteristic fix so I don't have to use custom modified version

philips77, 2024-10-10:
Hi,
I've found this issue: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/335 where it was requested for a connection to succeed despite bonding failure.
It's actually written here:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/ec3332bd88d32bde1de82815034f8c3d296dd5ef/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L344-L357

I have mixed feelings now, should the connection succeed, or not, had bonding been requested from the device side just after connection. But as this was already implemented, let's leave it for compatibility reasons.
This leaves us with the other issue - failing handling of the second bonding.

philips77, 2024-10-11:
OK, so I did some testing on 5 of my phones. I know why the request gets stuck. But I discovered few things I want to share.

You remember this:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/ec3332bd88d32bde1de82815034f8c3d296dd5ef/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2513-L2524 where the `onCharacteristicRead` is receiving 137 (bonding failed) error. I believe that it's enough to remove line 2515 (check for code 137) and let is pass to the next `else`.

However, similar callback for write - `onCharacteristicWrite` never gets 137. I tried all my phones, from Android 6 until 14 and no 137. I just get the "Bonding failed" from the broadcast receiver.

philips77, 2024-10-11:
I'll release fixes in 2.9.0-beta03 in few minutes. Please test :)

LightWeek, 2024-10-11:
I'll test tmrw or on Monday. Thx for the fix.

LightWeek, 2024-10-14:
Hi. I've been testing it for all day and the request does not stuck anymore. 
But there are some other issues, I'll report them separately.

PR #580 (2024-08-29): Bug fix: Cancel tasks when they are removed from the task queue
This PR fixes #552.

When the device gets disconnected while there are some tasks enqueued, the cancelled tasks will be notified and `.fail(...)` or `.invalid(...)` method will be called.

Also, all the tasks of a task queue (`AtomicQueueRequest` or `ReliableWriteRequest`) will be notified when they are removed from the queue (for example on disconnection, or when value validation fails during reliable write).
PR #597 (2024-10-11): Fixed handling cancelled bonding for read/write operations
This PR fixes #552, again.

Write and read requests trigger bonding when sent using an unsecured link to a protected attribute. If bonding fails, e.g. it's cancelled, there are 2 things happening:
1. Read characteristic and Read descriptor requests get status 137 in `onCharacteristicRead` callback. This will be ignored.
2. Write characteristic and Write descriptor requests don't get such callback. They don't get any callback, actually.

In both cases, the bond state broadcast receiver gets transition `BOND-BONDING` -> `BOND_NONE`. This [... cut]